FYSETC/FYSETC-SPIDER #4: In the spider I can use Pt100 direct

jjrricardo, 2021-05-17:
In the spider I can use Pt100 direct

GerogeFu, 2021-05-24:
I think that should be PT1000, you need a amplifier board to use PT100

aminshof, 2021-07-20:
hi, i have Trianglelab pt100 amplifier board, tried to connect with klipper
https://www.aliexpress.com/item/32829690782.html?spm=a2g0o.productlist.0.0.70321bcdaGnbPX&algo_pvid=125003df-e0cb-495b-95a2-8fb6896cdfdb&algo_exp_id=125003df-e0cb-495b-95a2-8fb6896cdfdb-14

here my config in extruder section :
sensor_type: PT100 INA826
sensor_pin: PC0
adc_voltage: 5.0

its read 52C when the room temp is 31C
there is 3.3v supply from the board, but the amplifier board need 5v, is there something i can do?

Thelm76, 2021-08-31:
> hi, i have Trianglelab pt100 amplifier board, tried to connect with klipper [...]

I think it is related with the issue in just opened here : #42 
I'm hoping someone can help me in a near future (maybe you found a solution since last month?)

Thelm76, 2021-09-03:
@aminshof I found a way to solve the issue (I have exactly the same configuration as you).
The easy way would be to plug your amplifier on a pin without pull-up resistor (I'll help you find one if you need it)
The hard way is to reach out the pull-up resistor just above the connector with a small soldering iron and remove it (with this solution you should power your amp board with 3.3V and leave "adc_voltage: 5.0")

closed 2026-02-04 by Rushmere3D