FYSETC/FYSETC-SPIDER #5: Bigtreetech mini 12864 v1.0 configuration Marlin with Fysetc Spider

karponline, 2021-05-31:
I had previously purchased a Bigtreetech mini 12864 v1.0 display. I want to connect it to the new Fystec Spider board. But only the backlight comes on, but the screen is blank. Is it normal for the speaker to make noises while loading the firmware? Will this display work with the new Fysets board?

alienboyxp, 2021-05-31:
https://wiki.fysetc.com/BIQU12864_On_Spider/ maybe this can be helpful

karponline, 2021-06-01:
Success. It works. I had to make three shifts:
1. Rotate Exp1 and Exp2 connectors 180 on BTT Mini
2. Swap the cables. Connect Exp1 from BTT Mini to Exp2 Fysetc Spider and Exp2 from BTT Mini to Exp1 Fysetc Spider
3. Remove R1 and C6 from BTT Mini

closed 2021-06-06 by karponline