FYSETC/FYSETC-SPIDER #6: Neopixel on Marlin 2.0.8.2 issue

alienboyxp, 2021-05-31:
Using Marlin 2.0.8.2 when Neopixel enabled using PD3 pin (following Spider scheme and wiki documentation) UART communication all low issues.

GerogeFu, 2021-06-01:
First we need to check if it is a hardware issue. Do you check other Marlin version or use Klipper ?

kolbenhans, 2021-06-05:
Same for me, NeoPixels doesn't work.

Using Klipper Firmware
5V voltage is available on LED strip

printer.cfg set up with:
`[neopixel caselight]
pin: PD3
chain_count: 20
color_order: GRB
initial_RED: 1.0
initial_GREEN: 0.0
initial_BLUE: 0.0`

I also tried to change the colors by SET_LED command as written in Klipper documentaton:
SET_LED LED=caselight RED=1.0 GREEN=0.0 BLUE=0.0

closed 2026-02-04 by Rushmere3D