FYSETC/FYSETC-SPIDER #8: TMC2209 V3.0 on Spider

oscaroecm, 2021-06-03:
I received TMC2209 V3.0 and pins are different compared to TMC2209 V3.1. Is there a way I can still use TMC2209 V3.0 on a Spider board or do I have tu use TMC2209 V3.1?

eliwong, 2021-06-09:
https://wiki.fysetc.com/images/2209%E4%B8%B2%E5%8F%A3%E6%8E%A5%E7%BA%BFV30.jpg
As the wiki shows, if you need to use 3.0 for SPIDER, we need to short-circuit the RX pin and SPREAD pin, and make sure that SPREAD is disconnected from the 2209 chip.
You can use a knife to cut the SPREAD, and then use a soldering iron to solder the SPREAD and RX together on the top of the module.
[image]

natnikos, 2022-04-28:
Have anyone tried this? i Have 5 fysetc TMC2209 v3.0 from my old board (SKR 1.4) that i want to use them to the new one ordered (Fysetc Spider v.2.2) and i don't want to destroy them (cutting pcb will).

dkuchay, 2023-04-02:
I also have this problem and seeing the date on the posts here it seems the vendor has no support for their own products and it would make sense to avoid these products until such time as fystec provides support

closed 2026-02-04 by Rushmere3D