FYSETC/FYSETC-SPIDER #10: How enable mosfet outputs

karponline, 2021-06-06:
I have a problem with PWM enabled mosfet outputs (1 for HotBed, 3 for Heat-End, 3 for fans). I checked with a meter and there is no voltage on any of them. The only thing that comes to mind is too low voltage to drive the mosfet because I use a 12V power supply. I was able to get all the stepper motors, limit sensors, temperature sensors and the display running. Everything works with this power supply. I also wonder how to properly connect the External MOS module V1.0

eliwong, 2021-06-09:
Spider's mosfet is driven by the onboard 12V (DC-DC), but the 12V circuit on the board requires a minimum input of 15V to work properly, so if you use a 12V input, you can mark it as "24V" and " The 12V" pin header is shorted.
[image]

karponline, 2021-06-10:
Thanks a lot for your help it works. The photos show my motherboard running during the first tests. Everything works very well. I have already ordered a new 24V power supply and I treat this solution as a test. I don't want to overload the mosfets with too much current. I recommend this product to everyone. 

[image]
[image]

closed 2021-07-11 by karponline