FYSETC/FYSETC-SPIDER #12: broken images

shiftingtech, 2021-06-07:
Several of the key diagrams are currently badly broken, both on github and the wiki.  For example, 
https://raw.githubusercontent.com/FYSETC/FYSETC-SPIDER/main/images/Spider_wiring.svg
currently renders in Chrome as
[image]

GerogeFu, 2021-06-09:
Try to download the whole project ,then check it in your computer

shiftingtech, 2021-06-15:
I'm not saying you're wrong, but you really shouldn't have to download the whole thing just to view the images.  Maybe you could post some JPEGs or something, just for ease of accessibility?

GerogeFu, 2021-06-16:
Ah, thanks for your reminding.

closed 2021-07-17 by GerogeFu