FYSETC/FYSETC-SPIDER #17: FAN not working on FAN1 (PB1) and FAN2 (PB2)

kees23, 2021-06-24:
Can't get fan to work on Spider board with FAN1 or FAN2.
With the cooling fan connected with FAN0 and below config, the fan works perfectly.

```
[fan]
pin: PB0
max_power: 1.0
kick_start_time: 0.5
off_below: 0.10

```
But when I connect the same fan to FAN1/2 and change the pin to PB1/2 it doesn't work.
No voltage on the pins. Either at 50% fanspeed or 100% fanspeed.
Using Klipper.

Something I have noticed is when I set the voltage jumpers to 12V, FAN1 is always ON. There is always 12V on the pins.
See attachment, the led at FAN1 is ON, immediately at startup. That doesn't look right.
Is this issue hardware related?

[image]

laserbeak, 2021-09-14:
Has anyone answered this?

kesor, 2021-10-08:
I have the same problem on my Spider/Klipper ... only FAN0 correctly works with 24V, the other two don't.

ichbaum, 2021-10-08:
I only have problems with FAN2.
0 and 1 works fine. All on 24V.
Fan2 won't start. Have testet several settings.

sleepingAwake, 2021-11-02:
So I too am having fan issues with my spider board purchased through the AliExpress fysetc store. 
First the boot loader it came with somehow didn’t allow usb or uart communication. Flashing the boot loader they have here got usb and uart working. 
Anyway the fans. 
Fan0 is connected to my hot end fan and jumpered to 12v. 
My print config is copied from here as well with some minor changes but the pins are correct from what I can tell. 
At first Fan1 wasn’t working. It’s my parts fan. 24v
Changing printer.cfg to below worked. I’ll tweak as needed.
[fan1]
Pb1

But Fan0 nothing.
I’ve tried various config changes and no voltage output. 

Also I’ve noticed when I apply some pressure around the fan0 connector I can hear a faint electric pop. It does it with the fans unplugged to. 
Could my board have an internal short?

sleepingAwake, 2021-11-03:
> So I too am having fan issues with my spider board purchased through the AliExpress fysetc store. First the boot loader it came with somehow didn’t allow usb or uart communication. Flashing the boot [...]

Fan0 may not be coming on due to normal behavior. Shame on me for not noticing the extruder fan behavior below 50c
But my board makes an electrical popping noise if I apply a small amount of pressure around the FAN0 connector and its mosfet. Sounds like the layers are shorting to each other. Its consistent and repeatable. Nothing is loose.

closed 2026-02-04 by Rushmere3D