FYSETC/FYSETC-SPIDER #21: exp1 and exp2 description

christian-hellwig, 2021-07-03:
the pinout of exp1 and exp2 are a little bit messed up. In some documents exp1 ist sd and exp2 lcd, sometimes the other way around. In the pinout graphics some ports are named totally wrong e.g. BA8 and DP2. MOSI is twice in the diagram whereas other ports are missing. Could you please provide one correct layout?

GerogeFu, 2021-07-14:
Sorry, i did not find BA8 in the pinout, could you point out that for me, give me screenshot. Thanks.

GerogeFu, 2021-07-14:
And no matter how the exp1 and exp2 decribe. You only need to stick to the wiring here. https://github.com/FYSETC/FYSETC-SPIDER/blob/main/images/mini12864-1.jpg

christian-hellwig, 2021-07-15:
I was referring to this image:
[image]
As you can see MISO is doubled on the right side, the notches on the EXP1 and EXP2 are pointing down, but on my Spider v1.0 and as well on my v1.1 the notches are pointing upwards. And I marked the mentioned BA8 and DP2 pins. On the v1.1 the EXP1 is on the left side but on my older v1.0 its on the right sight. Did you just exchange the markings or was the whole pinout changed?

GerogeFu, 2021-07-16:
I see we are working on this , BA8: PA8 DP2:PD2 , beside RST should be MOSI

GerogeFu, 2021-07-17:
Image updated.

closed 2021-07-17 by GerogeFu