FYSETC/FYSETC-SPIDER #22: No Connections

MartynJ970, 2021-07-04:
Hi have issues  as Uart and USB will not connect to Pi. 
Also wehn jumper is in 3.3v and BT0, It does not go into DFU mode, therfore I cannot connect to reflash st start again.
I think the USBC port is faulty. Please advise

dodasch, 2021-07-07:
I got the same problem, is your 3.3V LED on? On my Board only the 5V LED is on.

It doesn't matter whether I put the jumper between 3.3V and BT0 or not

I think the STM is dead.

ciro1138, 2021-07-07:
I am having this issue also. The only addition I have is there is no flashing LED by the SDcard when trying to upload the firmware.

dodasch, 2021-07-07:
@ciro1138 
you can upload the firmware or bootloader via sd-card? 
I can't upload any data because i have no connection. 
Neither with USB or SD.

ciro1138, 2021-07-07:
@dodasch 

I was not able until a few minutes ago. I started unplugging things and for me, the BTT 2209 drivers are blocking 3.3v as well as an end stop was in the wrong pin. now I can flash the firmware without the drivers in place. start by unplugging your drivers and see if your 3.3v light turns on. 

Only figured this out because I bought several spider boards and plugged in a fresh one out of the box. luckily nothing shorted... yet....

dodasch, 2021-07-07:
@ciro1138 
I have unplugged everything from my Spider Board. 
My printer was working till day X, the Pi was unable to connect to the spider board :( 
I think i got a defective Board.

ciro1138, 2021-07-07:
@dodasch What stepper drivers are you using? I am curious because the BTT ones will not let this board work for some reason and still investigating.

dodasch, 2021-07-07:
i'm using tmc2209 v3.1 from fysetc, they work normal

MartynJ970, 2021-07-07:
I am using the Fysect TMC2209 3.1 that came with the board.
I have also unpluged everything exept power to the board from MW PSU. Still no connection Via Pi, USB, UART. Non functoning sd card bootloader. All three leds on stating voltage in and distributed is live. 
Really dissapointing, My Fystect Prusa Bear is spot on!
I would suggest to Fysect that all or most faulty boards should be (foc to user) return to them for their QC/MUF & RD to determin and fix issues which i can only think must be small manufacturing issues. 5hrs Burn in testing time prior to packing & despatch.

MartynJ970, 2021-07-09:
Guys,
The soloution to the DFU is to unplug all apart from the power to the Board. It will then enter DFU. I have messages Fysect to update manual, but they need to find a proper fix as reconnecting is a pain!!!

dodasch, 2021-07-10:
To unplug everything will not solve my 3.3V and DFU problem.

ciro1138, 2021-07-10:
Still 3.3v issues with the drivers. Must be some faults or bridged connector somewhere. Moved back to BTT 1.4 TURBOS and klipper is communicating without any issues over USB and UART also works.

GerogeFu, 2021-07-14:
Is the 3.3v power led on?

> To unplug everything will not solve my 3.3V and DFU problem.

dodasch, 2021-07-14:
> Is the 3.3v power led on?
> 
> > To unplug everything will not solve my 3.3V and DFU problem.

https://github.com/FYSETC/FYSETC-SPIDER/issues/22#issuecomment-875360793

Darkwingduc, 2021-08-01:
I'm having the same issue I don't get flashing when I insert the sdcard with firmware.bin on it. I've tried uploading the original bootloader with stm32cubeprogrammer after moving the jumper and the lights are on but my computer doesn't recognize that anything is plugged in, and the programmer can't find the board either. I feel like this is a driver issue with windows 10 but have tried everything and still can't seem to get it to work. Does anyone know of a fix for this or what driver I may need.

GerogeFu, 2021-08-04:
Did you follow the steps [here](https://github.com/FYSETC/FYSETC-SPIDER/tree/main/bootloader) ?  Did you close BT0 and 3.3v?

Darkwingduc, 2021-08-04:
I did try that. I moved the jumper but when i plug it into my computer it is not recognized. If I go into device manager it does not show up. I tried different cables including the one that came with it but still nothing. I am going to get a STlink v2 to try and connect through that but dont know exactly what pins need to be connected.

Darkwingduc, 2021-08-06:
I was able to upload the bootloader with the STlink tool and it is up and running. I think this would solve the other issues as well. It's probably not as easy as just using usb but it really wasn't that hard, just 4 wires and you're connected. Now I just need to figure out why my z axis motors run at different speeds. Does this board support drv8825?

ciro1138, 2021-08-06:
@Darkwingduc 

Is there something you referenced for your fix I can try. I would like to try and use these boards since I can no longer return them.

Darkwingduc, 2021-08-06:
i bought an st link v2 from amazon. It connects near where the raspberry pi pins are. It uses the 3.3v, dio ,clk, and gnd pins and i was able to connect to the board using that and stm32cubeprogrammer. i marked the pins it goes to with dots. and here is a link to what it looks like. I got the cheap usb version and it worked great. [https://www.google.com/search?q=st+link+v2&rlz=1C1GCEA_enUS924US924&source=lnms&tbm=shop&sa=X&ved=2ahUKEwiojOS_xpzyAhXuUd8KHaaGBJYQ_AUoAXoECAEQAw&biw=1280&bih=577&safe=active&ssui=on](url)
[image]

beikeland, 2021-08-07:
> To unplug everything will not solve my 3.3V and DFU problem.

some boards come shipped with jumpers installed on the endstop headers, between 3.3v and GND - that'll effectively keep the board from working.