FYSETC/FYSETC-SPIDER #23: 3.3V does not work when stepper drivers installed, firmware also unable to flash

ciro1138, 2021-07-09:
I have purchased 2 spider 1.0 boards, both having the same 3.3v issue with stepper drivers plugged in. Initially, I could not figure out why the SD would not flash with a firmware.bin in the root directory. I realized 3.3v was not working after plugging in my second board with no stepper drivers installed. i initially tried with BTT 2209 driver boards, and have also tried now with Fysetc 2209 v3.1 boards and have the same 3.3v issue once installed.

Additionally, I am trying to flash the firmware for Klipper without a successful flashing no matter USB, uart, no boot, 32k, 64k and all combinations tried. I have checked my wiring for UART and tried the both firmwares from the github as well as make menuconfig downloaded from the raspi. I also tried to reflash the bootloader via stm32cube. I tried both bootloaders provided but no luck getting the connection to work. 

From klipper log:
cu 'mcu': Starting serial connect
Got error -1 in read: (11)Resource temporarily unavailable
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):

dodasch, 2021-07-10:
I get my second board now and its working. 
I used the same stepper drivers (tmc2209 v3.1 from fysetc).

ciro1138, 2021-07-10:
@dodasch then I was sent 2 bad boards.....

dodasch, 2021-07-12:
Support said to me, if i get a new Spider i should short circuit the board between 24+ and GND Input for a few seconds. 
But this will not help you with your sent boards :/

GerogeFu, 2021-07-14:
Check the issue analysis here : https://github.com/FYSETC/FYSETC-SPIDER/blob/main/Spider%203.3v%20issue.md

ciro1138, 2021-07-14:
@GerogeFu

ciro1138, 2021-07-14:
@GerogeFu 

I don't believe the drivers are damaged in any way because I am using them in a different board without issues. 

Regarding residual power on 24v it takes 10 or so seconds for the 24v to look like it is drained. I do want to note that when trying to flash my second board I jumpered 5v over USB and never plugged in 24v to try and connect to klipper. The firmware would not take on this board either trying no boot, 32, and 64 over USB. It never switched to cursor and in putty I cannot find a USB address. That is when I tried to connect via stm32 program to flash the bootloader and tried all available  versions. I did notice a list of sectors, can we get a photo of that from a known working board? I was suppr to see 6-8 sectors from my limited experience.

GerogeFu, 2021-07-15:
When i flashed bootloader, it look like this : 
[image]

beikeland, 2021-08-10:
> Regarding residual power on 24v it takes 10 or so seconds for the 24v to look like it is drained.

Problem is 24V feeds another Vm rail that is feeding stepper drivers though a diode; there is much more capacitance there and no load with no drivers installed - will keep charge much longer.

> That is when I tried to connect via stm32 program to flash the bootloader and tried all available versions. I did notice a list of sectors, can we get a photo of that from a known working board? I was surprised
> to see 6-8 sectors but my experience is limited
> in this program

found this on discord no need to use stm32cube (remove `:leave`and `-R` if you want to proceed to flash application firmware without re-entering dfu mode)
```
wget https://github.com/FYSETC/FYSETC-SPIDER/raw/main/bootloader/Bootloader_FYSETC_SPIDER.hex
objcopy --input-target=ihex --output-target=binary Bootloader_FYSETC_SPIDER.hex Bootloader_FYSETC_SPIDER.bin && dfu-util -R -a 0 -s 0x08000000:mass-erase:force:leave -D Bootloader_FYSETC_SPIDER.bin
```