FYSETC/FYSETC-SPIDER #24: Probe Problem

dodasch, 2021-07-11:
I use Klipper and have Problems with my Inductive Probe. 

My Probe is always Triggered if i configure NO and always closed if i configure NC. 

if the pin is configured to ^!PA3 (Normally Closed)
measured 3.3V between Pin PA3 & GND
Console:
QUERY_PROBE = probe: open

if the pin is configured to ~!PA3 (Normally Closed)
measured 2.64V between Pin PA3 & GND
Console:
QUERY_PROBE = probe: open

if the pin is configured to ^PA3 (Normally Open)
measured 3.3V between Pin PA3 & GND
Console:
QUERY_PROBE = probe: TRIGGERED

if the pin is configured to ~PA3 (Normally Open)
measured 2.64V between Pin PA3 & GND
Console:
QUERY_PROBE = probe: TRIGGERED

if i put metal under the probe, i can measure 24V between PA3 & GND and the LED on the probe lights up.  

Can you help me?

dodasch, 2021-07-11:
Here is my Klipper config:

https://pastebin.com/embdmx9J

photocromax, 2021-07-12:
I guess you do need a NPN, maybe there is a way to connect a PNP too, but not sure how.

dodasch, 2021-07-12:
Okay this Issue can be closed, but it should be marked on the wiring that you only can use a NPN Probe.

closed 2021-07-12 by dodasch

sbusso, 2021-08-07:
having the same problem, PA3 should provide 24V, did you solve your issue?