FYSETC/FYSETC-SPIDER #25: BL Touch / 3D Touch configuration

karponline, 2021-07-13:
What is the minimum BLTouch configuration in Marlin? I connected it according to the Spider 1.0 diagram (to PA0, PA2 and PA3). The power supply is correct 5V. I uncommented #define BLTOUCH and #define Z_SAFE_HOMING. But even Self-Test doesn't work. Does the microswitch have to be connected to PA0?

GerogeFu, 2021-07-14:
We have updated our wiring these days, you should wire the cable according to diagram.
[image]
Only the lateset Marlin bugfix-2.0.x here : https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32f4/pins_FYSETC_SPIDER.h, have the correct BLTOUCH pins define
[image]

glucz, 2021-08-01:
I can confirm that this is not working as described on the wiring diagram and set up in the config file. BLtouch gets power and does the initial self-test, but then nothing else is functional.

karponline, 2021-08-02:
@glucz it works but I didn't have time to write back. I downloaded this exact version. If you are fetching code via git you have to switch the branch to BUGFIX. Check if you have SERVO0_PIN defined - it is important.

spicker, 2021-08-06:
I have the same problem. BLTouch only does the self test but nothing else. It also looks like the wiring diagram is contradictory. As wired the bltouch is connected to PA2/PA0, but the red box next to the probe says PA3/PA0. It's the same on the 1.1 diagram.

glucz, 2021-08-06:
Well... I don't know. I have a 1.0 version and it didn't work with PA2. I think in my case the PA2 socket might have blown out when I drove the BLTOUCH with 24 volts ... I really have no idea why they would set up wiring in such a way where this is possible. There are a lot of 5V only ports that wouldn't even take up a sensor slot. Anyways, I had to use PA1 (X max) and now it's working.

So either the board came damaged, I damaged it by following the not-so-foolproof instructions or the Y max is not PA2 on V1.0

All in all, it was a $120 learning experience with the damaged board and the 2 blown BLTOUCH sensors.