FYSETC/FYSETC-SPIDER #27: Wrong labeling of VCC and GND

xenon2008, 2021-07-16:
Hi!

I am a newbie who is dealing with the Spider Board for the first time and is also building my first Voron!
But when I was about to wire up the RPi, I noticed something!
I wanted to connect 2x VCC and 2x GND from the Pi PWR & UART interface and it occurred to me that the pin on the top right is black, but was marked as 5V!
In the other section or the picture, however, this is again listed as GND!

Sorry i hope everybody here can understand / follow me what i mean, because my English Skills are so terrible Bed.
I am a Austrian User, and could only speak German =)

Some Pictures what i mean from the Documentation of the Spider 1.0 Wiring:

[https://raw.githubusercontent.com/FYSETC/FYSETC-SPIDER/main/images/Spider_v1.0_wiring.jpg](https://raw.githubusercontent.com/FYSETC/FYSETC-SPIDER/main/images/Spider_v1.0_wiring.jpg)
[image]

[image]

Kind Regards from Upper Austria ;)

photocromax, 2021-07-16:
Yes it's a mistake in the second picture. If you are worry, I can confirm that pin is GND.

GerogeFu, 2021-07-17:
Oh, that is mistake, we are right on it.

GerogeFu, 2021-07-17:
Updated.

closed 2021-07-17 by GerogeFu