FYSETC/FYSETC-SPIDER #29: 3.3v regulator overheating

CompaxChris, 2021-07-18:
I was going to test a spider board that just arrived but I did not get very far. There seems to be a short in the 3.3 voltage system somewhere, since the 3v3 regulator is overheating, and the voltage on the 3v3 line have droppet to 0.05v I can't locate where it is. I have added a image of the board setup and a thermal image of the 3v3 regulator.

I can find any other places on the board that seems to overheat unexpectedly.

[image]
[image]

dreamcat4, 2021-07-18:
its almost as if... there is a short somewhere due to a mis-wiring. like the 3v3 has been bridged to GND

interesting! please let us know if you find out what is going on here. so i can then also check on my own spider too. many thanks

CompaxChris, 2021-07-19:
So far I can't find it with thermal or with a microscope so it's probably a very "clean" short with low resistance. The last measure to find the short is destructive testing, cutting traces. The board will probably not be useful afterwards but it could still be interesting to find out where it is. Is the layout of the board available somewhere?

dreamcat4, 2021-07-19:
no i dont believe the pcb layout is available, and it seems to be a multi-layer board

have you tried changing the `USV` jumper in the middle of the board, and powering it from the usb-c port instead? does that give the same result? then it would seem to be a short. be sure you have everything disconnected, remove all the stepsticks and connecters etc. Disconnect everything apart from the usb cable

CompaxChris, 2021-07-19:
Yea without the layout its hard. If it's a four layer board the central layers are usually only GND and power layer. If 3v3 is used as the power layer it it's even harder. However if 5V, 12V or 24V is used as the power layer the cutting the 3v3 lines is at least feasible. However the black solder mask makes it quite had to spot the traces. A short like this is not uncommon in PCB production so the best option is probably just to get a new one. I was unlucky with this unit.

USB is 5V so it's still sent through the same 3v3 linear regulator. 

Fysetc followed up and is sending a new board.

closed 2021-07-19 by CompaxChris