FYSETC/FYSETC-SPIDER #34: SKR TFT 35 or TFT 70 Touch Screen with Spider

jojodroopy, 2021-08-02:
Hello, Is it possible to use a BTT SKR TFT35 or TFT70 type screen for choosing the Marlin or Graphic screen as on SKR Pro 1.2 or MKS L Gen L. I use the connector at the center of the Map with 5V, GND, RX1 and TX1. I have an inserted SD card back or remove but I do not have a dialog with the card. Thanks a lot for your help.

GerogeFu, 2021-08-04:
Should work with TFT screen, but i think you better to run Marlin firmware that BTT recommend. That TFT need some specific configurations.

jojodroopy, 2021-08-04:
Hello and thank you for your answer. I left on the basis of the configuration proposed by BTT. But can you tell me what are the specific settings for this type of screen?

GerogeFu, 2021-08-10:
Check "Marlin Dependencies"
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

jojodroopy, 2021-08-10:
Hello, ok for your information. But my screen works perfectly on a Turbo or other BTT card, but I can not operate it on the Spider. Is there a specific port to declare?

TheAlexPG, 2022-05-07:
> Hello, ok for your information. But my screen works perfectly on a Turbo or other BTT card, but I can not operate it on the Spider. Is there a specific port to declare?
[image]
Hello!I got this to work on my spider. I swapped the PX and TX wires at the entrance to the board. Serial port 1, serial port 2 -1. Only CR10_Stock_Display must be enabled