FYSETC/FYSETC-SPIDER #36: USB communication with Pi

Jhonno84, 2021-08-02:
Can't get my Spider V1.1 to communicate with my Pi via USB or UART (following the GitHub guide).. 

Flashes Bootloader and compiled firmware fine from STM32 Programmer but still won't talk to my Pi4.

OstapFerensovych, 2021-08-03:
Try to flash without bootloader. I had the same problem with Spider V1.1 and rpi4.
make menuconfig and choose Bootloader offset "No bootloader", and then flash klipper.bin to 0x8000000. Maybe it will help. Looks like bootloader that fysetc provide is a little buggy.

beikeland, 2021-08-07:
Bootloader only provides functionally to flash from SD card, for USB connection it is not needed - however getting support from the community probably assumes the bootloader is installed so that might be best to just find the correct offset 32k or 64k and it should work (unfortunately not possible to tell which one was installed at factory)

Jhonno84, 2021-08-12:
Thank you guys.. This was solved using 0 offset bootloader.

bakaufman, 2021-10-08:
> Thank you guys.. This was solved using 0 offset bootloader.

Still flashed by SD card with 0 offset? or using something else?