FYSETC/FYSETC-SPIDER #37: X+ endstop not firing on PA1

Recoil1980, 2021-08-05:
On setting up the new board I had sensorless homing setup.  I ran into issues not being able to properly set the sensitivity and decided to switch to mechanical end stops.  This is on an Ender 5 Plus modified for core XY, so end stops are at max X & Y (right rear).  Board connectors are wired with red wire on tan pins, black on black pins ([see picture](https://github.com/FYSETC/FYSETC-SPIDER/blob/main/images/Spider_v1.1_wiring.jpg))

X+ was on PA1 and Y+ was on PA2.  The X+ would not trigger but the Y+ would.  I swapped connections on the board and X+ on PA2 would trigger but Y+ on PA1 would not trigger. I put Y+ back on PA2 and moved X+ to PA3, and both trigger correctly.  The soldering on the back of the board on PA1, PA2, and PA3 looks good.  Is it possible there is an issue with my PA1 connection?

Darkwingduc, 2021-08-08:
I ran into the opposite issue with mine. It is most likely just an issue with the code. try different configs and see if something works. I had the opposite issue where in klipper the endstop postion was set to max like this (_position_endstop:280_) It kept homing backwards and I changed stepper direction and tried many other things until I finally noticed that one line was screwing me up. Maybe that one line is all you need.

GerogeFu, 2021-08-10:
> On setting up the new board I had sensorless homing setup. I ran into issues not being able to properly set the sensitivity and decided to switch to mechanical end stops. This is on an Ender 5 Plus [...]

If you have a multimeter, you can check the connection of the DIAG pin.
[image]

lvitol, 2021-10-05:
Related#49 , ran into similar issue when setup sensorless homing, found error in PinOut.SVG by check pin connection with multimeter