FYSETC/FYSETC-SPIDER #38: Klipper's flash-sdcard.sh script can't find the SD card.

jtbraun, 2021-08-06:
My board updates fine from the sdcard via the firmware.bin/old.bin method.

If I use Klipper's built-in flash-sdcard.sh script, though:
`./scripts/flash-sdcard.sh /dev/serial/by-id/usb-Klipper_stm32f446xx_140025000750305538333620-if00  fysetc-spider-v1`

This fails when it's trying to issue the first GO_IDLE_STATE command to the SD card.

Presumably the bootloader is doing something to the sdcard or the pin config after it checks for firmware.bin, preventing the remote access scripts from working.  I'd check the bootloader sources, but they aren't available. Reportedly this USED to work with the older bootloader...

jtbraun, 2021-08-06:
I've annotated the failures w/ extra print statements, but the core failure is the GO_IDLE_STATE command failing (OSError Exception marked w/ `******` at the start of the line).   Note that many of these exceptions in the normal script get eaten and rethrown as new ones, so the final `failed to mount SD Card` is a much more generic error message than `failed to reset SD Card`.
```
$ ./scripts/flash-sdcard.sh /dev/serial/by-id/usb-Klipper_stm32f446xx_140025000750305538333620-if00  fysetc-spider-v1
Flashing /home/pi/klipper/out/klipper.bin to /dev/serial/by-id/usb-Klipper_stm32f446xx_140025000750305538333620-if00
Checking FatFS CFFI Build...
Connecting to MCU...Connected
('BOARD CONFIG ', "{'baud': 250000, 'klipper_bin_path': '/home/pi/klipper/out/klipper.bin', 'cs_pin': 'PA4', 'klipper_dict_path': '/home/pi/klipper/out/klipper.dict', 'cd_pin': 'PB10', 'spi_bus': 'spi1', 'current_firmware_path': 'OLD.BIN', 'device': '/dev/serial/by-id/usb-Klipper_stm32f446xx_140025000750305538333620-if00', 'mcu': 'stm32f446xx'}")
Checking Current MCU Configuration...Done
MCU needs restart: is_config=1, is_shutdown=0
Attempting MCU Reset...Done
Waiting for device to reconnect...Done
Connecting to MCU...Connected
('BOARD CONFIG ', "{'baud': 250000, 'klipper_bin_path': '/home/pi/klipper/out/klipper.bin', 'cs_pin': 'PA4', 'klipper_dict_path': '/home/pi/klipper/out/klipper.dict', 'cd_pin': 'PB10', 'spi_bus': 'spi1', 'current_firmware_path': 'OLD.BIN', 'device': '/dev/serial/by-id/usb-Klipper_stm32f446xx_140025000750305538333620-if00', 'mcu': 'stm32f446xx'}")
Initializing SD Card and Mounting file system...
MOUNTING
INIT SD
TRY GO_IDLE_STATE
******OSError('flash_sdcard: failed to reset SD Card\nNote that older (Version 1.0) SD cards can not be\nhot swapped.  Execute FIRMWARE_RESTART with the card\ninserted for successful initialization.',)

SD Card Flash Error: Failed to Initialize SD Card. Is it inserted?
OSError('flash_sdcard: failed to mount SD Card, returned FR_NOT_READY',)
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1172, in main
    spiflash.run()
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1124, in run
    self.run_reactor_task(self.run_sdcard_upload)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1108, in run_reactor_task
    k_reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1087, in run_sdcard_upload
    self.mcu_conn.configure_mcu(printfunc=output_line)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 923, in configure_mcu
    "Failed to Initialize SD Card. Is it inserted?" + "\n" + repr(e))
SPIFlashError: Failed to Initialize SD Card. Is it inserted?
OSError('flash_sdcard: failed to mount SD Card, returned FR_NOT_READY',)
```

beikeland, 2021-08-07:
I updated to the 32k bootloader and see the same problem, cannot access SD card. Oddly enough reverting to the available 64k bootloader didnt help either. Having the sources would be nice indeed.

GerogeFu, 2021-08-10:
Try to change "scripts/spi_flash/board_defs.py" 
[image]

jtbraun, 2021-08-13:
That did get farther, GO_IDLE_STATE succeeded. It can't detect a filesystem, but perhaps the card is corrupted. I'll work at it a bit more to try to tell what's going on.

jtbraun, 2021-08-13:
Here's a dump of sector 0, which results in the fatfs library returning "FR_NO_FILESYSTEM".   There are repeating patterns of something like 0x1642e0 every ~19 bytes. The values change a little. I'll try rebooting.
```
CARD IS IDLE!
READ sector:0 count:1
DUMP 0x0000: 00 00 00 00 00 00 00 00 00 16 42 E0 00 00 00 00
DUMP 0x0010: 00 00 00 00 00 00 00 00 00 00 00 00 16 62 00 00
DUMP 0x0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 16
DUMP 0x0030: 62 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x0040: 00 00 16 62 40 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x0050: 00 00 00 00 00 16 62 60 00 00 00 00 00 00 00 00
DUMP 0x0060: 00 00 00 00 00 00 00 00 16 62 80 00 00 00 00 00
DUMP 0x0070: 00 00 00 00 00 00 00 00 00 00 00 16 62 A0 00 00
DUMP 0x0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 16 62
DUMP 0x0090: C0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x00a0: 00 16 62 E0 00 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x00b0: 00 00 00 00 16 82 00 00 00 00 00 00 00 00 00 00
DUMP 0x00c0: 00 00 00 00 00 00 00 16 82 20 00 00 00 00 00 00
DUMP 0x00d0: 00 00 00 00 00 00 00 00 00 00 16 82 40 00 00 00
DUMP 0x00e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 16 82 60
DUMP 0x00f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x0100: 16 82 80 00 00 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x0110: 00 00 00 16 82 A0 00 00 00 00 00 00 00 00 00 00
DUMP 0x0120: 00 00 00 00 00 00 16 82 C0 00 00 00 00 00 00 00
DUMP 0x0130: 00 00 00 00 00 00 00 00 00 16 82 E0 00 00 00 00
DUMP 0x0140: 00 00 00 00 00 00 00 00 00 00 00 00 16 A2 00 00
DUMP 0x0150: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 16
DUMP 0x0160: A2 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x0170: 00 00 16 A2 40 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x0180: 00 00 00 00 00 16 A2 60 00 00 00 00 00 00 00 00
DUMP 0x0190: 00 00 00 00 00 00 00 00 16 A2 80 00 00 00 00 00
DUMP 0x01a0: 00 00 00 00 00 00 00 00 00 00 00 16 A2 A0 00 00
DUMP 0x01b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 16 A2
DUMP 0x01c0: C0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x01d0: 00 16 A2 E0 00 00 00 00 00 00 00 00 00 00 00 00
DUMP 0x01e0: 00 00 00 00 16 C2 00 00 00 00 00 00 00 00 00 00
DUMP 0x01f0: 00 00 00 00 00 00 00 16 C2 20 00 00 00 00 00 00

SD Card Flash Error: Failed to Initialize SD Card. Is it inserted?
OSError('flash_sdcard: failed to mount SD Card, returned FR_NO_FILESYSTEM',)
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1177, in main
```

GitjoPowershell, 2021-09-11:
> Try to change "scripts/spi_flash/board_defs.py"
> 
> ```
>     'fysetc-spider': {
>         'mcu': "stm32f446xx",
>         'spi_bus': "swspi",
>         "spi_pins": "PA6,PA7,PA5",
>         "cs_pin": "PA4",
>         "current_firmware_path": "OLD.BIN"
>     }
> ```

This works nicely for my spider 1.1 with sd card inserted 
thanks a lot ! Way easier to update without putting voron v2 upside down ;)

beikeland, 2021-09-11:
Does changing this affect the other boards that use the same fysetc-spider entry? S6 among others. Solution works for spider, confirmed with several boards and sd card combinations.

falo2k, 2021-09-24:
Confirmed that this change works on a Spider v1.1 with a 32k bootloader connected over UART.

adelyser, 2021-12-29:
Has anyone tried this recently without the fix described here? There was a change recently that releases the SPI from the bootloader on STM32F4 processors and it should alone resolve this issue without needing software SPI.

falo2k, 2021-12-29:
Yep.  Klipper3d/klipper#4885 fixed it for me so the swspi change was no longer necessary.  If it still is for the Cheetah then that should use a new definition.

philgood, 2023-04-09:
I had the same issue, updated with the following spa_pins as suggested and pushed firmware via UART to the SD card and got a successful cash. 

Pull request [#4255 ](https://github.com/Klipper3d/klipper/pull/4255) was submitted May 2021 and fixed the issue, however it seems that the spider config has reverted back? 

I think another pull request is needed to resolve future spider board being flashed via UART to the SD card.

CodeMonkeyX, 2024-01-29:
Just an update. This [pull request](https://github.com/Klipper3d/klipper/pull/6443) worked for me, we just have to hope more people test it and get it merged into Klipper. Basically it adds an option (-s) to the "flash-sdcard.sh" script that sets the correct SPI Clock for our Spider boards (and any other boards that have this issue). If it gets merged then all we would need to do is add t he -s option when flashing, instead of making our files dirty every time.