FYSETC/FYSETC-SPIDER #39: Possible Issue in PinOut.SVG

xenon2008, 2021-08-08:
Good Morning,

I am planing to use one of the RGB Ports for a Chamber Fan on my Voron 2.4 which i am building currently.
Because i already had all 3 FAN Ports in use, i think over to use a RGB Port for a additional Fan (Chamber Fan)

I noticed that in the [PinOut.SVG ](https://raw.githubusercontent.com/FYSETC/FYSETC-SPIDER/main/Spider_V1.0_Pinout.svg) the Negative Pin is marked as Red? 
[image]

And in the [Wiring.SVG](https://raw.githubusercontent.com/FYSETC/FYSETC-SPIDER/main/Spider_v1.0_wiring.svg) i think that "+" & "-" are reversed?
[image]

I am not 100% sure, but if i understand this right "+" should be left and "-" right? - regarding to this Diagram here:
[image]

bigsubgit, 2021-08-08:
Hello, I'm starting now to studying the spider for my new 2.4, how could be a negative pole if there we put a jumper cap?

bigsubgit, 2021-08-08:
Sorry, perhaps I was wrong in my previous post.

GerogeFu, 2021-08-10:
All '+' is 12v/24v , and '-' is GND

xenon2008, 2021-08-10:
> 
> 
> All '+' is 12v/24v , and '-' is GND

Sure, but i mean the color on the first picture.... GND is RED?
and on the second Picture i think the in the Green & Blue Section "+" & "-" are reversed?

GerogeFu, 2021-08-11:
No, the middle point is not GND, it is common point for jumper, so if you can close a jumper on 24v and the middle point or on 12v and the middle point. That means you can select voltage.

GerogeFu, 2021-08-11:
> Hello, I'm starting now to studying the spider for my new 2.4, how could be a negative pole if there we put a jumper cap?

jumper cap

xenon2008, 2021-08-11:
> 
> 
> No, the middle point is not GND, it is common point for jumper, so if you can close a jumper on 24v and the middle point or on 12v and the middle point. That means you can select voltage.

oh sorry my Mistake!
But it is also right that on the two Blue Ports  the Voltage Pin is right?

GerogeFu, 2021-08-12:
Stick to this: all '+' is 12v/24v, and all '-' is GND. Then you won't make mistake.