FYSETC/FYSETC-SPIDER #40: "Can't write to TMC SPI "stepper_x/y/extruder register, GLOBALSCALER error"

Jhonno84, 2021-08-12:
Getting this error..

Example of my configs

[stepper_x]
step_pin: PE11
dir_pin: PE10
enable_pin: !PE9
microsteps: 16
rotation_distance: 40
endstop_pin: PB14  # PA1 for X-max
position_endstop: 0
position_max: 485

[tmc5160 stepper_x]
spi_bus: spi4
cs_pin: PE7
diag1_pin: PB14
run_current: 1
hold_current: 0.8
stealthchop_threshold: 0
interpolate: True

Tried different steppers in X leaving the others blank. This was working fine in my S6 V2. 

Trying to do anything with the steppers with TMC5160 drivers causes Klipper to shutdown. 

4 jumpers in the bottom positions on the driver slot, as per Spider Github.

Jhonno84, 2021-08-29:
I've tried a V1.0 5161 I have laying about, and that works. My others are V1.1 and they do not work.. There is a default SPI difference between the 2.. 
[image]

masterleo, 2022-11-05:
got the same issue but with nex drivers but on all axis
is it normal ? 
# This file contains common pin mappings for the BigTreeTech Octopus V1.
# To use this config, the firmware should be compiled for the STM32F446 with a "32KiB bootloader"
# Enable "extra low-level configuration options" and select the "12MHz crystal" as clock reference

# after running "make", copy the generated "klipper/out/klipper.bin" file to a
# file named "firmware.bin" on an SD card and then restart the OctoPus with that SD card.

# See docs/Config_Reference.md for a description of parameters.

## Voron Design VORON2 250/300/350mm BigTreeTech OctoPus V1 tmc5160 UART config

## *** THINGS TO CHANGE/CHECK: ***
## MCU paths                            [mcu] section
## Thermistor types                     [extruder] and [heater_bed] sections - See https://www.klipper3d.org/Config_Reference.html#common-thermistors for common thermistor types
## Z Endstop Switch location            [safe_z_home] section
## Homing end position                  [gcode_macro G32] section
## Z Endstop Switch  offset for Z0      [stepper_z] section
## Probe points                         [quad_gantry_level] section
## Min & Max gantry corner postions     [quad_gantry_level] section
## PID tune                             [extruder] and [heater_bed] sections
## Probe pin                            [probe] section
## Fine tune E steps                    [extruder] section

[mcu]
##  Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
##--------------------------------------------------------------------
serial: /dev/serial/by-id/usb-Klipper_stm32f429xx_250025001951303439373431-if00

restart_method: command
##--------------------------------------------------------------------

[printer]
kinematics: corexy
max_velocity: 300  
max_accel: 3000             #Max 4000
max_z_velocity: 15          #Max 15 for 12V TMC Drivers, can increase for 24V
max_z_accel: 350
square_corner_velocity: 5.0

#####################################################################
#   X/Y Stepper Settings
#####################################################################

##  B Stepper - Left
##  Connected to MOTOR_0
##  Endstop connected to DIAG_0
[stepper_x]
step_pin: PF13
dir_pin: !PF12
enable_pin: !PF14
rotation_distance: 40
microsteps: 32
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper
endstop_pin: PG6
position_min: 0
##--------------------------------------------------------------------

##  Uncomment below for 250mm build
#position_endstop: 250
#position_max: 250

##  Uncomment for 300mm build
#position_endstop: 300
#position_max: 300

##  Uncomment for 350mm build
position_endstop: 350
position_max: 350
[... 577 lines cut ...]

#   Use PRINT_END for the slicer ending script - please customise for your slicer of choice
[gcode_macro PRINT_END]
gcode:
    # safe anti-stringing move coords
    {% set th = printer.toolhead %}
    {% set x_safe = th.position.x + 20 * (1 if th.axis_maximum.x - th.position.x > 20 else -1) %}
    {% set y_safe = th.position.y + 20 * (1 if th.axis_maximum.y - th.position.y > 20 else -1) %}
    {% set z_safe = [th.position.z + 2, th.axis_maximum.z]|min %}
    
    SAVE_GCODE_STATE NAME=STATE_PRINT_END
    
    M400                           ; wait for buffer to clear
    G92 E0                         ; zero the extruder
    G1 E-5.0 F1800                 ; retract filament
    
    TURN_OFF_HEATERS
    
    G90                                      ; absolute positioning
    G0 X{x_safe} Y{y_safe} Z{z_safe} F20000  ; move nozzle to remove stringing
    G0 X{th.axis_maximum.x//2} Y{th.axis_maximum.y - 2} F3600  ; park nozzle at rear
    M107                                     ; turn off fan
    
    BED_MESH_CLEAR
    RESTORE_GCODE_STATE NAME=STATE_PRINT_END