FYSETC/FYSETC-SPIDER #41: Having issues installing Marlin onto the Fysetc-Spider

420AyanAli, 2021-08-13:
I have flashed by board with the new bootloader functions (_Bootloader-FYSETC_S6.hex_) using STM32CubeProgrammer, and all went well, but I am getting and error where PlatformIO gives me an error of the board _FYSETC_S6_8000_ being undefined: 

[image]

Am I being a complete idiot, or is this where I should be changing _FYSETC_S6_ to _FYSETC_S6_8000_
in the _platformio.ini_ file. According to this https://github.com/FYSETC/FYSETC-SPIDER#41-marlin
 
**But github says this:**
You need to enable following define in configuration.h file

#define MOTHERBOARD BOARD_FYSETC_SPIDER
default_envs = FYSETC_S6 (For old bootloader,boot address is 0x10000, see below)
default_envs = FYSETC_S6_8000 (For new bootloader,boot address is 0x8000, see below)
 
So upon searching for _default_envs_ in _Configuration.h_ I have not found it

I Believe the board is not damaged as when I bridge Boot0 and 3.3V the board goes into DFU mode and STM32CubeProgrammer connects successfully.

Since _FYSETC_S6_8000_ is not available, and I am getting an error I went back to the old bootloader _Bootloader-FYSETC_S6_10000_, I have compiled my code, and have uploaded through STM32CubeProgrammer using the old bootloader with _default_envs = FYSETC_S6_ instead.

When I exit DFU mode and try turning on the board, the board turns on fine, but windows does not detect the board as a device in device manager, and the board's LCD does not light up. I am using the pre-provided firmware from the Fysetc-Spider Github Repo, but have changed the LCD to RepRapDiscount Full Graphic Smart Controller.

Thank you in advance for any help you offer, I will try to reply with any more important information you may need as fast as possible!

GerogeFu, 2021-08-13:
Sorry, this should be my fault, i just update the ini. You can have a try now.

420AyanAli, 2021-08-13:
Alright I'll have a look now!

420AyanAli, 2021-08-14:
I've done some further testing, and the same problem is still there @GerogeFu

GerogeFu, 2021-08-19:
The version in our github is little bit old, please try Marlin upstream bugfix-2.0.x, i will fix this ASAP.

420AyanAli, 2021-08-20:
Okay, I'll try and set that up later today, and thank you

420AyanAli, 2021-08-20:
Alright everything has compiled successfully, I have uploaded the successfully compiled bootloader to the board using STM32CubeProgrammer, but once again the LCD does not turn on, nor does the board get detected by windows Device manager. Attached is my marlin folder with all my files and the success message:

[image]

My STM32CubeProgrammer configuration:

File path: C:\Users\ayana\Desktop\Marlin\Marlin-bugfix-2.0.x (Cantilever)\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts
Start Address: 0x08008000 (as I am using default_envs = FYSETC_S6_8000)

420AyanAli, 2021-08-20:
Alright everything has compiled successfully, I have uploaded the successfully compiled bootloader to the board using STM32CubeProgrammer, but once again the LCD does not turn on, nor does the board get detected by windows Device manager. Attached is my marlin folder with all my files and the success message:

[image]
[Marlin-bugfix-2.0.x.zip](https://github.com/FYSETC/FYSETC-SPIDER/files/7021398/Marlin-bugfix-2.0.x.zip)

My STM32CubeProgrammer configuration:

File path: C:\Users\ayana\Desktop\Marlin\Marlin-bugfix-2.0.x (Cantilever)\Marlin-bugfix-2.0.x\.pio\build\FYSETC_S6_8000\firmware.bin
Start Address: 0x08008000 (as I am using default_envs = FYSETC_S6_8000)

GerogeFu, 2021-08-28:
Did you change configuration.h file? The serial post should be -1.

420AyanAli, 2021-08-29:
Yes, I have changed it to -1

GerogeFu, 2021-09-03:
I test again. It works, you should upload bootloader https://github.com/FYSETC/FYSETC-SPIDER/blob/main/bootloader/Bootloader_FYSETC_SPIDER.hex. When uploading the firmware, you need to check the boot offset is right or not. 
[image]

GerogeFu, 2021-09-03:
I just update the Marlin firmware in our github.

420AyanAli, 2021-09-05:
Alright, I will have a look in a few hours, by boot offset do you mean the Start Address being 0x08008000?

GerogeFu, 2021-09-07:
Yes

> Alright, I will have a look in a few hours, by boot offset do you mean the Start Address being 0x08008000?

closed 2024-06-18 by 420Ayan420