FYSETC/FYSETC-SPIDER #44: TMC2208

msazanov, 2021-08-31:
Good afternoon, faced with problem drivers tmc2208 bt fysetc

https://aliexpress.ru/item/32848553467.html?spm=a2g39.orderlist.0.0.4b8b4aa68oAwrM&_ga=2.102432385.1279670221.1630433926-2024624615.1628585182&_gac=1.258551672.1628608684.CjwKCAjwx8iIBhBwEiwA2quaq1s9TrKFMzMXaR9MgGZCyCCvbFw1T3gEdR303A7xuWOkJI_QkSBpDhoC-oUQAvD_BwE

After setting up the configuration when testing the motors with the command
STEPPER_BUZZ STEPPER = stepper_x
Cooler hotend starts spinning 100%
And the printer falls out into an error:
Recv: // Unable to read tmc uart 'stepper_x' register IFCNT
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: // Klipper state: Not ready
Recv: !! Unable to read tmc uart 'stepper_x' register IFCNT

[klipper.log](https://github.com/FYSETC/FYSETC-SPIDER/files/7086101/klipper.log)
[printer.txt](https://github.com/FYSETC/FYSETC-SPIDER/files/7086138/printer.txt)
I also use Raspberry 3b + connected via UART

please help!!!

GerogeFu, 2021-09-01:
You need to solder the 3 pad to join them together.
[image]
 
[image]

msazanov, 2021-09-01:
[image]
[image]
[image]
I know about the jumper and it has been sealed for more than 3 years :) Previously, I used these drivers on the MKS GEN-L there was made a splitter with a resistor going to the UART pins (RX and TX) If you look at the Spider v1.1 circuit from the PDN contact (for example, motor x) there is one wire that goes directly to PD7, it seems to me that this may be the problem, but I'm not sure.

msazanov, 2021-09-01:
Found a solution! for the tmc2208 drivers from fysetc, the UART is on pin 5, when on the spyder PDN board pin 4. To fix this problem, it is enough to solder pins 4 and 5 from the driver together.
PS: on this driver pin # 4 is NC and is not connected to the chip
[image]
[image]
[image]