FYSETC/FYSETC-SPIDER #54: New board won’t connect

shadowdro, 2021-10-30:
I have a brand new board that has been in the antistatic bag until last night. I recently purchased the Voron 2.4 kit. I am unable to get it into dfu mode. I followed the instructions on the wiki.￼￼ I am unable to get it into DFU mode. Since the 32K boot loader supposed to be on the board since June I tried flashing the 32K uart via SD card.￼, which appears to have failed. And then tried doing the same for USB, and neither can connect to the pi.￼ right now when I put the firmware on an SD and put it onto the board it stays flashing for over 40 minutes. ￼after looking through these issue requests I tried clearing the power on the board and removing the stepper drivers and everything except for the two jumpers to enter DFU mode. It is still unable to connect and does not seem to respond via any commands. Please assist.￼

shadowdro, 2021-10-30:
[image]

RcTomcat1, 2021-10-30:
Same here. Just connected my board for the first time. Nothing. Tried with a Linux and Windows system, three different usb cables, 24 and 5v....no dfu ....

shadowdro, 2021-10-31:
￼￼someone in the Discord said they received a charge only usb cable. I'm going to try another usb cable and  see if that fixes B it.

shadowdro, 2021-10-31:
Tried new USB cable.... it connected... Kit came with charge only USB....

closed 2021-10-31 by shadowdro