FYSETC/FYSETC-SPIDER #55: Nevermore Micro V5 Duo Not working on E1 Out

laserbeak, 2021-11-03:
Hello,
I have a Nevermore Micro V5 Duo that isn't working on E1 Out. When I start it, it runs fun but eventually slows down and completely stops.
Here's my printer.cfg:

```
[fan_generic nevermore]
#  Nevermore - In E1 OUT Position
pin: PC8
max_power: 1.0
kick_start_time: 0.5
```

And here are my macros that I use to control it:
```
[gcode_macro nvmOff]
gcode:
    SET_FAN_SPEED FAN=nevermore SPEED=0

[gcode_macro nvmHi]
gcode:
    SET_FAN_SPEED FAN=nevermore SPEED=1

[gcode_macro nvmMid]
gcode:
    SET_FAN_SPEED FAN=nevermore SPEED=.5

[gcode_macro nvmLo]
gcode:
    SET_FAN_SPEED FAN=nevermore SPEED=.25
```

Does anyone have any ideas?

GerogeFu, 2021-11-04:
You can switch your E1 OUT to FAN0 if your FAN0 works to check if the board works, be careful that FAN0 max current is 1A, so your nevermore fan current should not exceed 1A

laserbeak, 2021-11-04:
Thanks for the reply. My hotend fan is connected to FAN0 and functions normally.  All fan ports are populated (Voron 2.4)

closed 2026-02-04 by Rushmere3D