FYSETC/FYSETC-SPIDER #58: many problem with Spider and bad support from FYSETC

quocanhcgd, 2021-11-12:
i have 2 board spider 1.1 
One of this work well but other don't work.
My board has many problems:
1. When i flash usb firmware . Pi not recorgnize. 
2. When i flash uart mod. Pi can connect it. But no heat extruder, no heat bed. I see led of heat not light up. use voltage meter no power out to extruder and bed.
3. I try contact fysetc on taobao but they don't know why and want me to write on github. no star for FYSETC support.
With me 30-40usd is not problem. But it can use to evaluate level of support. 
next time i will buy from bigtree and mks.

GerogeFu, 2021-11-12:
We checked the video you sent, and give you support, you can check the message.

GerogeFu, 2021-11-16:
Use no bootloader option when compile klipper and fix heat section in config works.

quocanhcgd, 2021-11-17:
Confirm usb can connect if use no bootloader. But heat still not work. I request change new board.

closed 2026-02-04 by Rushmere3D