FYSETC/FYSETC-SPIDER #61: only 2 out of 4 Z drives works with Marlin fw

Zooyork01, 2021-11-17:
I am uploading the marlin fw on a spider v1.1 which drives a voron 2.4 with 4 Z drive. 
In configuration_adv.h default: #define NUM_Z_STEPPER_DRIVERS 1 
By default only 1 driver is selected and it will run M3 driver 
If I choose 4 drivers then the error : No E stepper plug left for Z4!
If I choose 3 driver then only M6 M3 works.
I believe the issue is that all missing channels are configured as extruder and not as potential Z drive.

Is there a properly configured firmware exist for voron 2.4 Spider? Or do you have fix for this issue? Thanks in advance

GerogeFu, 2021-11-18:
Normally, we use Klipper firmware to run voron2.4. Don't have Marlin version yet, but if you want to run it, i think, you need to change `pins_FYSETC_S6.h` file
Change E1_* E2_* to Z3_* Z4_*
[image]

Zooyork01, 2021-11-18:
Thanks for the quick answer I tried and it works for first try! I have now 4 Z motors working. 

Where should I share my final firmware? I am building a voron 2.4 400x400 mm. Would be kind to the users to make it available.

closed 2026-02-04 by Rushmere3D