FYSETC/FYSETC-SPIDER #64: [heater_bed] heater_pin: PB4 not working

dodasch, 2021-11-24:
If i configure my bed to PB3, there is no LED or 24V working.
If i configure the bed to PB4, the LED is on and i have 24V. 

Is the PB3 Pin defect? Everything else works fine.

acidcoolxxx, 2022-01-20:
On my board the v2.2 the PB4 bin is Fan2 WTF

mrw464, 2022-01-30:
Yeah on my 2.v2, PB4 does not work for the BED OUT. Not sure what's going on there. I ended up using PB3 / the E2 OUT.

fmonaca, 2022-04-03:
You have to feed 24v to the BED IN port to have power going to PB4 and out to the SSR, or there will be no current to switch on PB4. You can have the 24v coming directly from the PSU or you can get it from the POWER of the Spider.

fmonaca, 2022-04-03:
> On my board the v2.2 the PB4 bin is Fan2 WTF

Are you using it on a Voron? The printer.cfg for a Voron usually links FAN2 (controller fan) to the bed activation, so that the skirt fans turn on as soon as the bed is on.

Laroto, 2023-01-05:
> You have to feed 24v to the BED IN port to have power going to PB4 and out to the SSR, or there will be no current to switch on PB4. You can have the 24v coming directly from the PSU or you can get it from the POWER of the Spider.

What if I am using DC bed heater? I want to use a 100W DC heater but no idea on how to wire it

netweaver1970, 2023-01-05:
wiring example for from the Fysetc site ... look at the black heater connection, that's a DC heater. 

> What if I am using DC bed heater? I want to use a 100W DC heater but no idea on how to wire it

wiring examples on the Fysetc wiki ... look at the black heater's connection, that's the DC heater. https://wiki.fysetc.com/Spider/#31-spider-wiring