FYSETC/FYSETC-SPIDER #65: Unable to communicate with TMC2209 3.1 on X/Y-MOT only (spider 2.2)

alexnagelberg, 2021-11-29:
I have everything the same as the wiring diagram and printer.cfg on wiki and even swapped the TMC2209 chips around to see if they were dead for any reason but they work fine on the others. When I communicate with E[0-4]-MOT and Z-MOT they all work just fine -- same jumper, same driver installed. I get this error for X and Y-MOT trying DUMP_TMC on klipper: `Unable to read tmc uart 'stepper_x' register GCONF` and `Unable to read tmc uart 'stepper_x' register IFCNT` when trying STEPPER_BUZZ.

Using included v3.1 TMC2209 drivers with spider v2.2 board and klipper (printer.cfg off wiki, pins matching pinout diagram)

GerogeFu, 2021-11-29:
How many jumpers on your X and Y stepper driver slot?

alexnagelberg, 2021-11-29:
One for the PDN -- same as the wiring diagram on wiki and the Z and E MOTs.[image]

alexnagelberg, 2021-11-29:
Turns out it needs a jumper on the 24v. No idea what that connector was until someone on discord helped. Could you please update the 2.2 docs to make a note on the top right header for those two motors?

GerogeFu, 2021-11-30:
Try to replace the jumper on the socket downside. Sometime it affect. And how many jumpers did you use for the stepper driver?

GerogeFu, 2021-11-30:
There is no room for this note, you can check our Trident kit diagram here https://github.com/FYSETC/FYSETC-Voron-Trident/blob/main/VORON_Trident_Spider_v22_Wiring.png

alexnagelberg, 2021-11-30:
That yellow box is perfect. That guide was certainly not the easiest to find but hopefully someone else struggling will come across this and see it. Thanks!

geaz, 2022-01-12:
I was also struggling with this, because I use a 12V PSU and thought this info is not relevant for me. Turns out it is ....

kkmvoron, 2022-09-15:
I did the same like as you and got same error. Do you find out solution ?

closed 2026-02-04 by Rushmere3D