FYSETC/FYSETC-SPIDER #66: TMC 'stepper_x' reports error: GSTAT: 00000004 uv_cp=1

skfab, 2021-12-01:
I am getting this error on my Voron2.4 Spider 1.1 with 2209 3.1 Fysetc stepsticks, After starting a print I will get it on ether the x or y axis. I have replaced the x and y 2209s with new ones but no change. Any help would much appreciated. 

TMC 'stepper_x' reports error: GSTAT:
00000004 uv_cp=1 (Undervoltage!)

skfab, 2021-12-02:
I changed the stealthchop_threshold from 1  to 0 and raised the current settings, it seems to be working now.

closed 2021-12-02 by skfab

bogeyf14, 2021-12-19:
I am having the same issue on my spider v1.0 with tmc 2209 3.1. I also have stealthchop_threshold set to 1 and motor current is set to about 85% of rated current

dawidmpunkt, 2022-03-11:
> I am having the same issue on my spider v1.0 with tmc 2209 3.1. I also have stealthchop_threshold set to 1 and motor current is set to about 85% of rated current

Because I was also having this issue: A faulty stepper cable also causes this error. Try another stepper cable.

warrenfalk, 2022-06-27:
I received this same error a few days ago due to what turned out to be a bad 2209 (for anyone else coming here googling this error message). I ended up swapping slots with a working 2209 just to confirm the issue and then replaced it with a new one and the problem is resolved. The 2209 had worked fine for a couple years before giving out.

P1P1Monster, 2023-03-27:
Same situation, Cable problem or stepper driver broken both may cause this issue

WaschBaer0395, 2024-01-13:
same issue but its neither the stepper driver ( tested ) nor the cable ( swapped with a different one ) or the motor ( also swapped with a motor not throwing this error ) 
this only happens during homing sensorless sometimes and only way to fix it is to restart the whole printer

petercabra, 2024-03-18:
I am having the same issue, happens randomly when homing sensorless, mostly on Y axis (voron 2.4r2). Only way to fix it is to fix it is restarting the whole printer.