FYSETC/FYSETC-SPIDER #67: Can't connect over USB

OzzieOrca, 2021-12-02:
I can't seem to get my Raspberry Pi 3 Model B+ to connect to the Spider v1.1 board. Here's a list of things I've tried so far:

1. `ls -l /dev/serial/by-id/` returns `ls: cannot access '/dev/serial/by-id/': No such file or directory`
2. I copied `~/klipper/out/klipper.bin` to `firmware.bin` on the Raspberry Pi's SD Card, inserted that SD card in to the spider and powered the spider on. The Flash LED blinked quickly for 10s of seconds and then became solid. I've since tried that a couple times.
3. Before inserting the SD card the first time, I think the Flash LED blinked quickly on power up. After flashing and removing the SD card, the LED seems to stay off.
3. When in DFU mode (jumper between `3.3V` and `BT0`), `dfu-util --list` returns
    ```bash
    dfu-util 0.9
    
    Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
    Copyright 2010-2016 Tormod Volden and Stefan Schmidt
    This program is Free Software and has ABSOLUTELY NO WARRANTY
    Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
    ```
4. When in DFU mode, `dfu-util -R -a 0 -s 0x08008000:leave -D ~/klipper/out/klipper.bin` returns
    ```bash
    dfu-util 0.9
    
    Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
    Copyright 2010-2016 Tormod Volden and Stefan Schmidt
    This program is Free Software and has ABSOLUTELY NO WARRANTY
    Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
    
    dfu-util: Invalid DFU suffix signature
    dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
    dfu-util: No DFU capable USB device available
    ```
5. `dmesg -wH` returns a few different thing depending on how the board is powered and DFU mode:
    a. Non-DFU mode powered by 24v, no messages
    b. DFU mode powered by 24v
    Googling `error -110` seems to often indicate a power issue
    ```bash
    [Dec 2 02:02] usb 1-1.3: new full-speed USB device number 5 using dwc_otg
    [ +15.240212] usb 1-1.3: device descriptor read/64, error -110
    [ +15.360226] usb 1-1.3: device descriptor read/64, error -110
    [  +0.219991] usb 1-1.3: new full-speed USB device number 6 using dwc_otg
    [ +15.220212] usb 1-1.3: device descriptor read/64, error -110
    [Dec 2 02:03] usb 1-1.3: device descriptor read/64, error -110
    [  +0.120108] usb 1-1-port3: attempt power cycle
    [  +0.659938] usb 1-1.3: new full-speed USB device number 7 using dwc_otg
    [ +10.540067] usb 1-1.3: device not accepting address 7, error -110
    [  +0.100008] usb 1-1.3: new full-speed USB device number 8 using dwc_otg
    [ +10.540063] usb 1-1.3: device not accepting address 8, error -110
    [  +0.000114] usb 1-1-port3: unable to enumerate USB device
    [Dec 2 02:09] usb 1-1.3: new full-speed USB device number 9 using dwc_otg
    [ +15.240071] usb 1-1.3: device descriptor read/64, error -110
    [ +15.359972] usb 1-1.3: device descriptor read/64, error -110
    [  +0.219957] usb 1-1.3: new full-speed USB device number 10 using dwc_otg
    [Dec 2 02:10] usb 1-1.3: device descriptor read/64, error -110
    [ +15.360039] usb 1-1.3: device descriptor read/64, error -110
    [  +0.120173] usb 1-1-port3: attempt power cycle
    [  +0.659846] usb 1-1.3: new full-speed USB device number 11 using dwc_otg
    [ +10.540021] usb 1-1.3: device not accepting address 11, error -110
    [  +0.100000] usb 1-1.3: new full-speed USB device number 12 using dwc_otg
    [ +10.540000] usb 1-1.3: device not accepting address 12, error -110
    [  +0.000128] usb 1-1-port3: unable to enumerate USB device
    ```
    c. DFU mode and powered by USB 5v (jumper moved to `U5V`)
    ```bash
    [Dec 2 02:14] usb 1-1.3: new full-speed USB device number 13 using dwc_otg
    [  +0.290086] Under-voltage detected! (0x00050005)
    [  +4.160004] Voltage normalised (0x00000000)
    [ +10.740004] usb 1-1.3: device descriptor read/64, error -110
    [ +15.360089] usb 1-1.3: device descriptor read/64, error -110
    [Dec 2 02:15] usb 1-1.3: new full-speed USB device number 14 using dwc_otg
    [ +15.220092] usb 1-1.3: device descriptor read/64, error -110
    [ +15.360029] usb 1-1.3: device descriptor read/64, error -110
    [  +0.120084] usb 1-1-port3: attempt power cycle
    [  +0.669995] usb 1-1.3: new full-speed USB device number 15 using dwc_otg
    [ +10.530033] usb 1-1.3: device not accepting address 15, error -110
    [  +0.100077] usb 1-1.3: new full-speed USB device number 16 using dwc_otg
    [ +10.539932] usb 1-1.3: device not accepting address 16, error -110
    [  +0.000091] usb 1-1-port3: unable to enumerate USB device
    ```
    d. Non-DFU mode powered by USB 5v
    ```bash
    [Dec 2 02:17] Under-voltage detected! (0x00050005)
    [  +4.160008] Voltage normalised (0x00000000)
    [Dec 2 02:26] Under-voltage detected! (0x00050005)
    [  +6.239996] Voltage normalised (0x00000000)
    ````

It seems like the board is responding correctly to USB connections. I've tried a couple different cables that work with my phone (I can get my phone to show up in `dmesg`).  All the appropriate power LED are on and it's able to power the Pi when it's hooked up that way. Do you guys have any other ideas of things to try? Is something wrong with my board?

Thanks for your help! Hope I can figure this out :)

OzzieOrca, 2021-12-03:
I flashed a build to use UART today and configured the Pi for UART (per https://github.com/FYSETC/FYSETC-SPIDER/blob/main/firmware/Klipper/Connect%20RPI%20uart.md). `ls /dev/ttyAMA0` seems to work but Klipper still can't connect. 

My Klipper `printer.cfg` has
```
[mcu]
serial: /dev/ttyAMA0
restart_method: command
```

`klippy.log` has:
```
mcu 'mcu': Starting serial connect
webhooks client 1967506616: New connection
webhooks client 1967506616: Client info {'program': 'Moonraker', 'version': 'v0.7.1-194-geda5623'}
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed
```

GerogeFu, 2021-12-03:
Try to use no bootloader option https://github.com/FYSETC/FYSETC-SPIDER#1-boot-address-no, and flash it with command `dfu-util -R -a 0 -s 0x08000000:leave -D ~/klipper/out/klipper.bin`

OzzieOrca, 2021-12-03:
Somehow I got `dfu-util --list` working. The only thing I can think of is I unplugged the display cables EXP1 and EXP2. The only other thing is maybe powering the Pi from the Spider. I had a separate USB power running to the Pi at one point to keep it on.

After that, I did need to flash to the `0x08000000` address and overwrite the bootloader to get it working. Thanks for that! I've got a few things to figure out but it's connecting now :) Thanks!

closed 2021-12-03 by OzzieOrca

beikeland, 2021-12-09:
>  `dfu-util -R -a 0 -s 0x08000000:leave -D ~/klipper/out/klipper.bin`

Can still make use of `make flash FLASH_DEVICE=0438:df11` when _no bootloader_ is selected in `make menuconfig` which imho is a cleaner approach as it will work with updating also by make flash (but passing the /dev/serial/by-id/... device node as FLASH_DEVICE like `make flash FLASH_DEVICE=/dev/serial/by-id/usb-Klipper<fill inn the reset>`

But as the user is also mentioning UART using _no bootloader_ will force the user to use USB and BT0 jumper for any future updates. I would strongly suggest recommending restoring the bootloader to enable flash-sdcard.sh to work or sticking with just using USB.