FYSETC/FYSETC-SPIDER #73: 5v RGB (NEO-PIXEL/WS2812) pin out

Caskia, 2021-12-29:
I only find one pinout in the board for 5v rgb, is there any way to extend two independent pinout?
board version: spider v1.1
[image]

ejhoness, 2021-12-30:
Hi. hope this can help you. 
[image]

another RGB port with selectable voltage
and I believe you can use these other PWM ports with mosfet

sakoula, 2022-01-27:
Hi, I am having the same issue. I would like to connect a 5V Neopixel  `WS2811` ledstrip to some port other than the 5V RGB one.

I will power the ledstrip from an external 5V PSU. The only option that supports 5V is the FAN ports. Should I just go on and connect the Signal port to the V+ of the fan port?

Thanks!