FYSETC/FYSETC-SPIDER #74: Burned spider 2.2

RcTomcat1, 2022-01-05:
Hi,

One of the MOSFETs (at least I assume it it one) above the fan1 port just burned out on my spider 2.2
[image]

Is that the one connected to the fan1 port? Struggling to find the fault here and it would really help to know where to look.
Measured the fan1 parts cooling fan and could not find a short.
Same goes for the hotend cooling fan.

leanghoun, 2022-01-06:
Same exact one just burned for me as well. Above FAN1. All properly set to 24V. Brand new Spider 1.1, just going through initial setup. Would like to know if there are any warranties.
[image]

RcTomcat1, 2022-01-06:
Also building a voron from the Fysetc voron kit?

avongil, 2022-01-07:
Oh great.  I have had this happen to my V1.0 board.  Figured it would have been fixed by now :(

avongil, 2022-01-07:
I moved mine over to the RGB pins to the right.  Have a look at my config, they seem to operate a bit different so I had to mess with the pwm frequency of all the fans.   I guess I am going to stay away from Fan0, 1 and 2 connectors. https://github.com/avongil/VoronAxG68K/blob/main/printer.cfg

leanghoun, 2022-01-07:
> Also building a voron from the Fysetc voron kit?

Ah, not a Fysetc kit. A self-sourced Trident. I was really excited about the Spider board too. 

> I moved mine over to the RGB pins to the right. Have a look at my config, they seem to operate a bit different so I had to mess with the pwm frequency of all the fans. I guess I am going to stay away from Fan0, 1 and 2 connectors. https://github.com/avongil/VoronAxG68K/blob/main/printer.cfg

Yup, I've moved mine over to the RGB pins, but even those are behaving weirdly. The pin assignments for klipper are not correct, and while R/G pins are okay, the B pins are always on. I'll check your config for PWM adjustment. Thanks!

maralb1970, 2022-01-08:
Fysetc didn't add any backlash diodes over the fan outputs. I think it is a good idea to add these in future revisions.  I am building a Voron 2.4 too with a spider V2.2. After reading too many of these messages I will add a backlash diode (1N4001 or something like that) externally, probably crimping these along with the wires on JST crimp terminals.

Another thing not quite related....I hope there will be decent support for 4-wire fans in the future on controllers and firmware. Too many issues with PWM-ing 2 wire fans which all behave differently when controlling these by PWM....

avongil, 2022-01-10:
[image]

Spider V1.0 - happened immediately after a power up.

avongil, 2022-01-10:
> 

That's a good idea I think. I will do that as well as power the Pi4 from an external supply to not stress the 3.3v bus.

bofh, 2023-01-20:
I got my bed output burned on Spider V2.2. Bed is always on now (heating w/o limits, PCB itself is also heating)

Several questions:
* could it be caused by supplying bed voltage from a separate PSU? (Both do have electrically connected grounds and cases, voltage measured by multimeter between all points is nominal, was running in this configuration for a month or so no problem)
* could it be caused by wiring load (work area LED strip) to extruder output and turning printer on w/o driver installed? I've noticed temperature running away after this change.
* any way to prevent bed heating w/o limits in MOSFET short scenario?

I'm going to resolder MOSFET (it heats and looks deformed a bit). Any recommendations? Maybe different part or add some diodes?

egorcik, 2023-07-31:
Hi. 
Have you solved problem with bed always on?

bofh, 2023-08-02:
I've added external MOSFET module and used PD3 from RGB connector to drive it.