FYSETC/FYSETC-SPIDER #80: FAN0 is always ON

Laker87, 2022-01-24:
Hello. I have Spider 2.2 board and latest Marlin firmware.  Part cooling fan is connected to FAN0 and it's always ON, I can't control it. MOSFET seems OK.

jaschkopf, 2022-02-01:
Same problem here. Also Spider 2.2 Board, bought from Aliexpress FYSETC Official Store. Also the BED Output is always on. BED and FAN0 LED's are instant on after power up.

askkostya, 2022-02-04:
Same problem on Klipper firmware on Spider 2.2

jaschkopf, 2022-02-04:
Hey guys, I found the issue for the problem:

With Spider Version 2.2 they changed the Fan0 to Pin PA13 and FAN1 to PA14. The reason why PA13 is always on is, that it is double used, it's attached to FAN0 and the SWDIO PIN. So while booting the pin is pulley "high" and the FAN0 powers on. But with Pin PA13 you can control it normally. 

I recommend you to use FAN0 only for hotend-fan or connect it directly to 24V. If you reset your board (or an error occurs and the board resets by itself!!!) FAN0 is the only Fan that will have power. If the hotend is hot and the fan fails, your extrudermount will melt!

The reason why my bed heater was always on was a hardware issue. I found a cold solder joint at U5 Pin 9 (D-IN at CD4504B-EP Level shifter for the heaters). After resoldering it, the problem was fixed.

rusviktorger, 2022-06-11:
> With Spider Version 2.2 they changed the Fan0 to Pin PA13 and FAN1 to PA14. The reason why PA13 is always on is, that it is double used, it's attached to FAN0 and the SWDIO PIN. So while booting the pin is pulley "high" and the FAN0 powers on. But with Pin PA13 you can control it normally.

You're right, I bought two identical boards (Spider V2.2) at different times.
Fan0 (PA13) is also always on.
But that's really bad.
Can't change that somehow, maybe via the bootloader,
an older version maybe?

[German]
Du hast recht ich habe zwei gleich Boards (Spider V2.2) auch zur unterschiedlicher zeit gekauft.
Da ist Fan0 (PA13) auch immer an.
Das ist aber richtig schlecht.
Kann mach das nicht irgendwie ändern vieleicht über den bootloader,
eine ältere version  vieleicht ?

TinkerBridget, 2022-06-15:
Definitely something not ok with the 3 controllable fan sockets.

I can't run PA13 at 5v anymore, if I use it for the 24v fans they run all the time now, but not full power.  They used to run from powering up until Klipper started when they would stop and wait for commands, but something happened and now it is doing this.  I haven't bothered to test the pin output with various jumper settings and conditions as I'm busy trying to tune in a 2in1 out hotend.

Currently using PA14 at 24v for part cooling which seems to work, and PB2 at 5v for hotend cooling.

Unsure of the cause.