FYSETC/FYSETC-SPIDER #86: TMC 2209 v3.1 Unable to read tmc uart 'stepper_x' register IFCNT

Trytas, 2022-03-11:
Hi,

I'm at the moment trying to get my Voron running.
I've got the Spider v2.2 with 7 TMC 2209 v3.1 drivers installed.

I installed every jumper as shown in the instruction.
I Already tried to switch the cables to the steppers and also switch the drivers.

My next try is pulling my 2208s out of my CR-10 as soon as it has finished the print.

Here is my Config.
I just converted the .cfg to .txt
[config.txt](https://github.com/FYSETC/FYSETC-SPIDER/files/8231764/config.txt)

Trytas, 2022-03-11:
I now tried the "STEPPER_BUZZ STEPPER=stepper_x" Command in the console for all motors.

All 4 Z-Stepper and the Extruder are running without issues, but the Error is coming up for both gantry Steppers.

So not only the X but also the Y-Stepper.

gogsi1, 2022-04-04:
I have the same problem, were you able to solve it?

Kamilus2410, 2022-09-28:
Could you show how you done it? I have the same problem but that big black jumper is installed

GerogeFu, 2022-09-29:
Try this : https://github.com/FYSETC/FYSETC-SPIDER#52-tmc2209-connection-issue