FYSETC/FYSETC-SPIDER #87: Bed, Fan0 and Fan1 always ON

DrTriD, 2022-03-11:
Bed, Fan0 and Fan1 always ON state on Spider V2.2 when the board powered. I am not able to switch to OFF position.

I am using Marlin FW 2.0.9.3.

Does any one has a solution for that?

dckostin, 2022-04-25:
This was reported as a bug on the Marlin page {[BUG] Marlin 2.0.9.3 cooling fan #23418} and closed with a "fix" implemented in the latest bugfix-2.0.x branch.  However, I am now unable to start the print cooling fan at all (Fan0) and the backlight on my Mini12864 display will not turn on using the bugfix-2.0.x branch.  There is a new bug {[BUG] Bad fan control on MKS Monster8 V1.0 #24080} on the Marlin page that I think (hope) will turn into resolving these follow-on bugs.

dckostin, 2022-04-27:
I was able to fix my print fan by using the Marlin-bugfix-2.0.x release and then changing the TIMER_SERIAL in "buildroot\share\PlatformIO\variants\MARLIN_FYSETC_S6\variant.h" file which says:

#define TIMER_SERIAL TIM7 // TIMER_SERIAL must be defined in this file

By changing this to TIM4 resolved my problem.

See Marlin issue:  [BUG] Bad fan control on MKS Monster8 V1.0 #24080

Abeperspiration, 2024-06-09:
I have near same issue on my Spider v3.0. FAN0 is alway ON with klipper.