FYSETC/FYSETC-SPIDER #91: New board, unable flash firmware via SD, USB via SSH, USB via DFU both to PC & Pi

kullwarrior, 2022-04-01:
I got a new board, tried to flash multiple different approach without avail.
1) loading SD card with firmware.bin does not result in any blinking lights after turn on device. The SD card file remained unchanged.
2) Plugging USB type C board with 5v jumper with U5V (I get 3.3V & 5V lights) connecting it to Pi or PC does not get pick up by either device with or without 3.3V jump with BT0. I've also tried with 24V, 5v-DC5V jumper, BT0-3.3V jumper (I get 3.3V, 5V, 24V lights) without success. And finally I have swab type C cables with all other ones (all of them can transfer data between my PC and phone so I don't believe this is the issue).
I also did the firmware without any stepper, jumper (minus 5V to U5V or DC5V, and 3.3V-BT0)
Do I have a DOA board? I don't believe my pi is the problem as SKR E2 mini V2 does get pickup with the same USB socket, and the PC socket picks up any other USB devices.

megasqrt, 2022-04-23:
i have exactly the same problem

kullwarrior, 2022-04-28:
So update for any people who encounter this:
1) The USB type C's TX and/ or RX was faulty; this prevented serial connections via PC for STM32CubeProgrammer, serial connection with RPi. I'm unsure why the UART also failed
2) USB TTL will not work if you try to plug it into the four pin USB JST socket. At best you can use it as a 5V plugin to power the board.
3) I was able to recover SD firmware flashing by USB ST-Link V2 (https://www.amazon.ca/CANADUINO-Compatible-Circuit-Programmer-Debugger/dp/B07B2K6ZPK) You will have to use STM32  ST-Link Utility because the firmware for the ST Link on the board is outdated for STM32CubeProgrammer.
4) For some reasons my RPi does not like 32KiB bootloader, as soon as I flash my replacement board using a version without bootloader offset the RPi was able to pick up the board via USB serial connections.

closed 2022-04-28 by kullwarrior