FYSETC/FYSETC-SPIDER #94: Flashed Firmware, Screen (mini12864) now Dark

runnersigh, 2022-04-19:
I flashed the firmware via STM32 Programmer. I used the Klipper-32k-usb.bin to the address 0x8000000. Now when I disconnect the stm programmer and take the jumper off the 3.3v-BTO pins and reboot the board the screen is totally dark. Is this normal? 

Just to be clear I don't have anything hooked up to the board (no drivers, no fans) only the 24v and the screen (mini12864).

teeedubb, 2022-04-20:
Try flashing the no bootloader version - I had similar issues and that resolved it for me.

runnersigh, 2022-04-20:
What address did you use to flash the no bootloader version? Also what was
the name of the version that worked for you? I've tried several different
of the pre built versions and the screen is always dark.

dckostin, 2022-04-25:
I have the same issue with the dark screen.  The latest release version of Marlin (2.0.9.3) has an issue with the print cooling fan always running ([BUG] Marlin 2.0.9.3 cooling fan #23418).  The "fix" in the bugfix-2.0 branch turns the fan off, but now the mini12864 display light will not turn on (and the print cooling fan will not start).  I have confirm multiple times that returning to the Release branch 2.0.x will turn the light back so this is not a hardware issue but some code change in the bugfix-2.0 branch.