FYSETC/FYSETC-SPIDER #99: Missing example config folder for RRF

mak0t0san, 2022-06-02:
In the RRF readme (https://github.com/FYSETC/FYSETC-SPIDER/tree/main/firmware/RRF) in "Step4: config" , it says:

> There is config folder beside this README file, it is just an example config to run RRF

However, there is no config folder.

It would be helpful to have an example config folder, especially having the `board.txt` file.

jaysuk, 2022-06-02:
Use the config tool at https://teamgloomy.github.io/Configurator

mak0t0san, 2022-06-02:
I should have clarified that I'm trying to run a Spider v1.0 (which uses STM32F446). The configurator has the option for a Spider 407 board. I've got RRF running on the board, but there's no documentation for the SPIDER-RRF-WIFI module so it's a bit difficult to know which pins are for espDataReadyPin, TfrReadyPin and espResetPin.  

This is what I came up with so far for my board.txt file, which the configurator isn't generating:

```
//WiFi pins
8266wifi.espDataReadyPin = C.11
8266wifi.TfrReadyPin = D.1
8266wifi.espResetPin = A.8
8266wifi.csPin = C.9

// RX/TX Settings
8266wifi.serialRxTxPins = { C.7, C.6 }
serial.aux.rxTxPins = { A.10, A.9 }
```

This is the board.txt file that is generated from the configurator:
[image]

GerogeFu, 2022-08-30:
Uploaded

jaysuk, 2022-10-11:
ah, we (teamgloomy) don't support 446 based boards so you'd have to just
work it out with the schematic.
What I would say is don't bother. There has been no long term RRF support
on 446 chips after the initial build that they pushed out and its buggy at
best.

Thanks,

Jay

closed 2026-02-04 by Rushmere3D