FYSETC/FYSETC-SPIDER #100: Board does not respond to UART or USB

eljefe80, 2022-06-08:
Hi,

I have a Spider v2.2 that is not working. I have tried UART and USB connections on multiple raspberry pis and multiple linux machines. No usb device appears in lsusb. The board was functional and working well, then stopped responding midprint. I can program it successfully with an st-link v2, but cannot connect to it. Please advise.

Thanks,
eljefe80

Nugabar, 2022-06-09:
Please check with lsusb while jumper is betweeen EO-3.3V.
If it shows something like SAM xxxx in DFU mode, please try to write bootloader first, then firmware.
I solved problem like above.

eljefe80, 2022-06-11:
Do you been BT0 and 3.3V? for DFU mode? Yes I have tried that. I've also made sure that the jumper is on USBV5 to draw power from the USB connection

rainerschulte, 2023-08-06:
Icurrently ran into the same issue now. 
I have connected the Spider 2.2 to a Raspi 4B via USB. The Spider is recocnized by `lsusb` and "successfully flashed" with Jumper between BT0 and 3.3V. After unplugging Jumper and rebooting the Spider, I get `xw1170@spider:~/klipper $ ls -l /dev/serial/by-id
ls: cannot access '/dev/serial/by-id': No such file or directory
` UART will not work on my Pi4 because I have added a Fan to it wich blocks the Pins for the provided UART-Cable. I also tried UART on a PI Zero, but it gets no Connection. 
What's wrong here?

JD13xs, 2023-08-07:
How did you configure it in your build of the firmware; usb or Uart? You
can not do both ( at least to my knowledge)....

rainerschulte, 2023-08-07:
For Pi 4 I built Firmware for USB, since I cannot connect UART-Cable and Fan in the Housing. So I flashed the Spider with USB online "make flash..." and repeated later with SD-Card. Both were not showing a USB-Serial by-id. I tried to renew the Bootloader with STM Code-Programmer. When connecting USB to my Computer with Juper on Spider BT0 and 3.3V I had a connection and successful flash. 
When using the SD-Card to flash USB-Firmware to Spider, I noticed the blinking LED behind USB-Port. After that, the SD-Card file has changed to "old.bin". So I assume, I did everything right this time. But I still get no /serial/by-id.

The same Procedure I tried on Pi Zero with UART. But flashing Firmware for Spider with UART, following the Manual I also edited cmdline.txt and config.txt.

rainerschulte, 2023-08-07:
**_ok... finally solved!_** 
I reflashed the Bootloader again, and used the basic .hex file instead of the ...1000.hex. The SD-Card was flashed to 32kb Bootloader. so the 1000 is mentioned for 64k Bootloader. 
In the End I now have Klipper online and no Error Messages. A Fan is connected, a Thermistor for Hotend is connected and the Display is working. 
So I'm fine for today! Now I need to order some Parts for my new Voron 2.4r2!!

JD13xs, 2023-08-07:
Ive noticed that sometimes you have to run make / flash more than once to
work correctly.

closed 2026-02-04 by Rushmere3D