FYSETC/FYSETC-SPIDER #102: STM32CubeProgrammer won't flash bootloader

TinkerBridget, 2022-07-03:
14:48:38 : Time elapsed during the read operation is: 00:00:00.031
  14:48:45 : Memory Programming ...
  14:48:45 : Opening and parsing file: Bootloader_FYSETC_SPIDER.hex
  14:48:45 : Error: Parsing fail
  14:48:45 : Error: In line 38 :
  14:48:45 : Error: parsing count, expecting hex digit, found ' '

Not sure what the problem is here, but I will add it to the list of issues.  I just wanted to reflash with a newly generated klipper firmware to make sure that wasn't the problem and I can't seem to do it.

TinkerBridget, 2022-07-07:
I ended up not using the bootloader, compiled an updated klipper firmware and flashed it from my PC rather than using the SD card.

Board works, but it should have been possible to reflash the provided bootloader to it.

Meffesino, 2022-10-12:
I have got the same problem. I'm not able to use STM32CubeProgramming to flash the latest 32bit Bootloader.

ataman78, 2022-11-23:
Problem is 100% in your .hex. You must download not only .hex by "save as" but a whole project in zip. Then will work. Good luck)

Terecon, 2022-11-30:
> Проблема на 100% в вашем .hex. Вы должны скачать не только .hex с помощью «сохранить как», но и весь проект в zip. Тогда будет работать. Удачи)
Я прошивал не один раз, но сейчас не работает. Посмотрите видео. Что не так.
https://disk.yandex.ru/i/E_vNhVVvQbynjg