FYSETC/FYSETC-SPIDER #104: Klipper „usb to can bus bridge mode“ not possible with fysetc bootloader

RedFlash49, 2022-07-30:
Hi

I wanted to try the new usb to can bridge mode (https://www.klipper3d.org/CANBUS.html#usb-to-can-bus-bridge-mode) with my Spider 1.1 (with attached CAN transceiver) but only got USB errors on the pi with dmesg.
Now i replaced the fysetc bootloader with the bigtreetech octopus bootloader and the board is recognized and the CAN bus works.
The only disadvantage i found is that flashing with SD card is not possible anymore.
Could you please check your 32kB bootloader. I would prefer to use the right bootloader.

GerogeFu, 2022-08-11:
Try this one
https://www.dropbox.com/s/0zckknbnuz2tsjy/Bootloader-Spider_L.bin?dl=0 
Tell me if it works on your side

RedFlash49, 2022-08-13:
> Try this one
> https://www.dropbox.com/s/rxobica829fon36/Bootloader_Spider.hex?dl=0
> Tell me if it works on your side

Unfortunely not. Still „not accepting address, error -32“ with dmesg.

GerogeFu, 2022-08-15:
Screenshot that part of dmesg for me. I will have a check

RedFlash49, 2022-08-16:
[image]

RedFlash49, 2022-08-20:
Hi
It was an Error in the Klipper Firmware. I updated Klipper and flashed all MCU's to the newest Klipper Firmware Version and now it works with the fysetc bootloader.
To be shure i flashed the original fysetc bootloader and the firmware again and it still works.
Weird, that it worked with the other bootloader.

Thanks

RedFlash49, 2022-08-20:
.

slaesh, 2023-07-31:
@RedFlash49 do you have a display too? and might answer my quersion [here](https://github.com/FYSETC/FYSETC-SPIDER/discussions/132) ? =)

or maybe @GerogeFu do you know this?

RedFlash49, 2023-07-31:
HiYes i have a display buts its a hdmi display connected to the raspberry.I had the spider (1.1) connected with CAN and changed back to USB. There is no advantage with CAN and i suspected its causing too much traffic on the bus resulting in „timer too close“ errors.