FYSETC/FYSETC-SPIDER #111: Unable to read tmc uart 'stepper_x' register IFCNT

kaikaimax, 2022-10-23:


Sark1tama, 2022-11-25:
Same problem, did you solve it?

SpookyPDX, 2023-01-11:
check that the 24v/48v jumper is in place, chased this for few days, my issue was the jumper

vogodesign, 2023-01-15:
[image]

Has the problem been resolved?
I am also struggling with this issue.

SpookyPDX, 2023-01-20:
if only X and Y are effected its the 24v jumper at the end of the row of steppers, its the big jumpper in the bag with all the small jumpers

xrmatic, 2023-02-15:
Thanks! This fix it for me too.

[image]