FYSETC/FYSETC-SPIDER #118: TMC5160 Sensorless Homing Won't Work

kazooless, 2023-01-22:
I can't get senseless homing to work. I have Fysetc branded TMC 5160's and a Spider 2.2 board. All 4 jumpers are on per the wiring diagram. To follow is my Klipper config for the X & Y. The carriage moves in the right direction, but when it hits, it keeps trying.

Please help

##  Connected to X-MOT (B Motor)
[stepper_x]
step_pin: PE11
dir_pin: PE10
enable_pin: !PE9
rotation_distance: 40
microsteps: 64
full_steps_per_rotation:400  #set to 400 for 0.9 degree stepper
endstop_pin: tmc5160_stepper_x:virtual_endstop
position_endstop: 0
position_min: 0
position_max: 350
homing_speed: 20   #Max 100
homing_retract_dist: 0
#homing_positive_dir: true

[tmc5160 stepper_x]
spi_bus: spi4
cs_pin: PE7
diag1_pin: ^!PB14
run_current: 1
driver_SGT: -64

##  Connected to Y-MOT (A Motor)
[stepper_y]
step_pin: PD8
dir_pin: PB12
enable_pin: !PD9
rotation_distance: 40
microsteps: 64
full_steps_per_rotation:400  #set to 400 for 0.9 degree stepper
endstop_pin: tmc5160_stepper_y:virtual_endstop
position_endstop: 350
position_min: 0
position_max: 350
homing_speed: 20   #Max 100
homing_retract_dist: 0
#homing_positive_dir: true

[tmc5160 stepper_y]
spi_bus: spi4
cs_pin: PE15
diag1_pin: ^!PB13
run_current: 1
driver_SGT: -64

Print3DWorld, 2023-01-30:
Did you put the jumper in the diag slots for X and Y? Otherwise, you need to use the TMC driver comm tool to check if it's reading properly.

kazooless, 2023-02-11:
Thank you. That's what it turned out to be. I missed it on the diagram (easy to do when looking at it).

closed 2023-02-11 by kazooless