FYSETC/FYSETC-SPIDER #125: Board doesn't work. 3.3v led doesn't turn on

daniel-sousa-me, 2023-03-31:
I have the Spider connected to a Raspberry Pi, but I can't get it to work in any way.

A serial device shows up when connected through USB, but I can't do anything with it.

The 3.3V led never turns on. I have tried following the instructions on https://github.com/FYSETC/FYSETC-SPIDER/blob/main/Spider%203.3v%20issue.md with no success.

I've tried flashing with MicroSD (no SD led). I tried to flash the bootloader, but dfu-util doesn't see it.

Is the board dead? Is there anything else that I could try?

jakobwowy, 2023-04-01:
I have the same issue today. The board was running and the 3.3 led was ok. After some restarts and changing USB-Port -> Not connection to the printer and also no 3.3 led burning

I have a older version V1.0

1300371, 2024-10-29:
Same here: just plug step drivers and the 3.3V LED never come back.

@GerogeFu , at this time, my board is dead also 😢 ?

closed 2026-02-04 by Rushmere3D