FYSETC/FYSETC-SPIDER #130: Fan's not working

whoisjako, 2023-05-18:
Hello,

i got a Fysetc Spider 2.3 as an upgrade for my ender 3 with klipper. So far everything worked fine, my motors are running, i can heat nozzle and bed, but i'm unable to get any fan to work.

So what i  tried:

I wanted to use Fan0, Fan1 and Fan2 for the part cooling , extruder cooling and one for the electronics.
i have different fan voltages, so i swapped the jumpers to 24V or 12V corresponding to the fan. I updated the printer.cfg as following:

[heater_fan hotend_fan] 
pin: PA14
max_power: 1.0
fan_speed: 1
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0

[fan]
pin: PB2
max_power: 0.4
kick_start_time: 0.5
off_below: 0.10

[heater_fan controller_fan]
pin: PA13
kick_start_time: 0.5
heater: heater_bed
heater_temp: 45.0

I tried to use only one fan at first but however i'm swapping fans or voltages, the fans wont work. 
The leds at the fan connectors are lightning correctly when they are activated, just wont get any

What am i missing? tried different fan's already that all worked before i swapped my electronics..

Thanks in advance!

jako

closed 2023-05-18 by whoisjako

wanderlust-vlad, 2023-12-30:
I'm also facing such issue and have found no solution to date. Could you pls clarify?

skyguy94, 2023-12-30:
I had to reverse the polarity of the fan wires. After examining the board and the schematic it dawned on me that the red and black pins in the connector were reverse of what the board expected them to be. I just popped out the pins on the connector and swapped them.

wanderlust-vlad, 2024-01-01:
Thaat worked for me as well, thanks a lot!