FYSETC/FYSETC-SPIDER #137: Spider 3.0 documentation?

greenaar, 2023-07-08:
Is there a timeline for the spider 3.0 documentation?  In particular the differences in wiring between the 3.0 and the 2.3, the changelog, and perhaps most interesting, the differences in wiring the CanBus connections?

Thank you

RobTS, 2023-08-19:
Just in case it helps you, I just replaced my old setup with the Spider 3.0 and integrated the SB2209. I followed read up on how the Octopus does it, then used the pins suggested for the 2.3, works flawlessly now, just add the new CAN Bus ID to your `[mcu]` entry, wire everything up, and you're set!

[image]

oldmate45, 2024-09-17:
A year + later and still not much info out there...

cyberjak2k, 2024-09-17:
> A year + later and still not much info out there...

There's more than enough documentation to make a functional printer though I made a completely custom system using various things centred around this board and there was more than enough for me to make a fully functioning Core XY system. What were you hoping for that was not there?