FYSETC/FYSETC-SPIDER #138: Unable to read tmc uart 'stepper_x' register MSCNT

Safandula86, 2023-07-08:
Hi everyone, I have an issue with my newly assembled printer. I have checked the connections multiple times and couldn't find any issues with the wiring. When I try to perform Home All or Home X/Y, I receive an error message after reaching the 0,0 position: 
Unable to read tmc uart 'stepper_x' register MSCNT
sometimes:
Unable to read tmc uart 'stepper_y' register MSCNT

I have already connected pins 4 and 5 on each driver, but it didn't help. Do you have any suggestions?

Printer is Voron 350 on Spider v3, S2209 v4, Orange Pi 4

Kvis-Dev, 2023-08-20:
Hello. Have almost the same issue here. 
`Unable to read tmc uart 'stepper_z' register IFCNT`

Were you able to solve it?

Safandula86, 2023-08-21:
Yes, please check jumpers under the stepper drivers

Kvis-Dev, 2023-08-31:
My issue was different. Jumpers were fine, but I needed to solder a jumper between RX and TX pin on my drivers. See image below. 
**Read the documentation to your drivers before doing this.** 
[image]