FYSETC/FYSETC-SPIDER #143: Spider V2.2/klipper sensorless homing issue

Cabinfourus, 2023-10-04:
I am trying to configure Klipper to run my Cartesian 3D printer. I am having trouble with the diag pin for sensorless homing and the TMC2209 driver. Klipper keeps giving me this error " pin virtual_endstop used multiple times in config". I have been through the configuration multiple times and cannot see the conflict. The X and Y pins on the Spider V2.2 board share pins PB13 and PB14 with the X and Y min endstop connections. I want to use sensorless homing. When I add the diag pin configuration to my printer.cfg file, it comes back with the error. How do I configure the Spider V2.2 board with the Fysetc TMC 2209 V3.0 drivers to use sensorless homing in Klipper?
Also, I'd like to use the ADXL345 sensor with the Spider V2.2 board but I am not sure which pins to use for the connections.

Cabinfourus, 2023-10-04:
WOW! I own several Fystec boards and I am starting to use them but with the lack of help I am finding with this problem from them, I think I will go back to Makerbase or BTT boards.

ejhoness, 2023-10-05:
I don't know if I can help,
the x and y endstops are the same pin
``
[stepper_y]
step_pin: PD8
dir_pin: !PB12
enable_pin: !PD9
microsteps: 16
rotation_distance: 40
endstop_pin: tmc2209_stepper_x:virtual_endstop``

Cabinfourus, 2023-10-08:
This has been one of the most confusing boards I have ever dealt with. I will NEVER buy another Fyetc product.

Rushmere3D, 2023-10-10:
Both your X and Y are configured for endstop_pin: tmc2209_stepper_x:virtual_endstop & diag_pin: PB13

Simply looking at the provided pin out could fix the issue. Can't see how this is a Fysetc problem, more user error.

Rushmere3D, 2023-10-10:
And an ADXL sensor connects to the Raspberry Pi, not the spider. Follow the Klipper docs.