FYSETC/FYSETC-SPIDER #144: Tmc2209 Unable to read tmc uart 'stepper_z

MakkeLeon2, 2023-10-19:
Hello I have a problem again. The board arrived and I got it working. I did calibrate the Z endstop and the probe everything was working fine. I did change the fan connection to PB2-24v 2x controller fans and PB7-5v leds on the hotend. And still everything was working fine.Then I start heating the bed and the hotend and run QGL still everything ok. Then I run clean nozzle and after that I got { Unable to read tmc uart 'stepper_z' register IFCNT
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown}  this again. ps it did clean the nozzle ok.
I changed the driver Z already but that didn't help
[moonraker (2).log](https://github.com/FYSETC/FYSETC-SPIDER/files/13043053/moonraker.2.log)
[klippy (5).log](https://github.com/FYSETC/FYSETC-SPIDER/files/13043054/klippy.5.log)
[printer.txt](https://github.com/FYSETC/FYSETC-SPIDER/files/13043078/printer.txt)

MakkeLeon2, 2023-10-24:
I commented out uart_addres:0 but that did not change anything. I did change z(z-mot) to (E4-mot) PC5 PE0 PE1 PD11 and now I get (Unable to read tmc uart 'stepper_x)
Put in a new driver in x but that did not help. But if I say home it goes up a bit so Z works again before I changed it to E4 not

rafaelmontserrat, 2023-11-27:
Hi, did you figure this one out? I am having the same issue but I can't seem to solve it.

MakkeLeon2, 2023-11-27:
You can use STEPPER_BUZZ STEPPER=stepper_x or y or z1 or z2 or z3 not. In terminal To check the drivers. If it is only Z you can change it to E4 but you have to change the pins.I think the problem is no earth on socket.I had 3 times the board replaced.Every time the nossle hits the plate it burns out the board.I did buy this now one but that doesnot arrived jet so still no printing.I hope this one is better secured.

BIGTREETECH Manta M8P Control Board running Klipper with CB1/CM4 × 1
China / CB1+Manta M8P V2.0+8PCS TMC2209

$98.29 

Discount

B8585FBB366F

-$4.00

Subtotal

$94.29

Shipping

$0.00

Taxes

$0.00

Total

$94.29 USD

rafaelmontserrat, 2023-11-27:
Hey. Yeah... this board... is far from good from what I'm experiencing. I have the issue in ALL my drivers, X Y Z and E so yeah, need a fix but i guess i'll wait for fysetc support to come back to me...