FYSETC/FYSETC-SPIDER #149: Spider v2.3 flash firmware and bootloader - jumpering for dfu-util

tekgrrl, 2024-01-17:
We just had a failure of our trusty Spider v1.1 (our fault) and we're replacing with a Spider v2.3. 

The new board came with a base install of Marlin so I built Klipper on the Raspberry Pi planning to install it with `dfu-util`. For everything up to Spider v2.2 the instructions are to jumper BT0 and the 3.3v which are clearly marked on the board. I can't find these anywhere on the board or on any online diagrams of the board.

**How  do I use `dfu-util` with the Spider v2.3?**

Just to add, I tried using the SD Card and it seemed to flash OK (`firmware.bin` renamed to `old.bin` ) but the PI and the Spider wouldn't talk, even though there was a valid USB serial device. While trying different options with the Klipper firmware build I likely overwrote the bootloader and don't know how to restore that without using dfu-util. (this is likely a separate issue).

JD13xs, 2024-01-18:
They are there. Download the  pinout and wiring diagrams.
The location has changed. I have most of the revisions of the boards. Make
sure to change the pins on the fans as they are different.

tekgrrl, 2024-01-19:
Thanks, found it. It's marked as Boot0 on the silk screen on the back of the board.

closed 2024-01-19 by tekgrrl